powsybl/powsybl-core #1944: Increasing BranchResult

annetill, 2022-01-18:
* **Do you want to request a *feature* or report a *bug*?**

A feature.

* **What is the current behavior?**

In a `BranchResult` we can monitor active and reactive flows.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

After the base case and after each contingency, it would be great to monitor also the tap position of a transformer, that could changes if the transformer voltage control is active.

Following the same principle, a `ShuntCompenstorResult` has to be created to monitor reactive flow at connection terminal and the section count, that could change if the shunt voltage control is active.

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

annetill, 2022-01-18:
https://github.com/powsybl/powsybl-open-loadflow/pull/427

sylvlecl, 2022-01-18:
Indeed, more information could be very interesting:
for example, in other tools, we also get some information about the generators that reached their limit.

We could also add some information for phase shifters.

sylvlecl, 2022-01-18:
Some functional/design questions to be discussed:

1. Do we consider those new results as a new "category" of results which we could name "regulation results" ?
I think it would be relevant, because it's a little different from the existing branch / bus results which are just "state" result, not results about what has acted

2. Do we want to get those results:
    - always for all regulations
    - for all regulations, only if a boolean option has been activated
    - for all regulations of a given type, only if a boolean option has been activated
    - only for regulations we asked for by their ID

alicecaron, 2026-10-28:
Due to inactivity for more than 3 years, this issue is now closed. If you plan on solving the issue, please feel free to re-open it.