powsybl/powsybl-core #1950: Support for severity in functional logs

zamarrenolm, 2022-01-24:
* **Do you want to request a *feature* or report a *bug*?**
Request a *refactoring* in functional logs API to define Severity as a *Type* and LOG, WARN, ERROR, ... as *Values* of that type.
Discussed with @flo-dup.

* **What is the current behavior?**
Current functional logs API defines the different severity levels LOG, WARN, ERROR, ... as *Types* for `TypedValue` objects. 

In `TypedValue` class we have:
```java
public class TypedValue {
    public static final String UNTYPED = "UNTYPED";
    public static final String TRACE_LOGLEVEL = "TRACE_LOGLEVEL";
    public static final String DEBUG_LOGLEVEL = "DEBUG_LOGLEVEL";
    public static final String INFO_LOGLEVEL = "INFO_LOGLEVEL";
    public static final String WARN_LOGLEVEL = "WARN_LOGLEVEL";
    public static final String ERROR_LOGLEVEL = "ERROR_LOGLEVEL";
    public static final String ACTIVE_POWER = "ACTIVE_POWER";
```

A user of the functional log API creates typed value constants to set a severity for a functional log report entry. An example can be found in UCTE importer.

The class `UcteReportConstants` declares the following constants for typed values:
```java
public final class UcteReportConstants {
    private UcteReportConstants() {
    }
    public static final TypedValue TRACE_SEVERITY = new TypedValue("UCTE_TRACE", TypedValue.TRACE_LOGLEVEL);
    public static final TypedValue DEBUG_SEVERITY = new TypedValue("UCTE_DEBUG", TypedValue.DEBUG_LOGLEVEL);
    public static final TypedValue INFO_SEVERITY = new TypedValue("UCTE_INFO", TypedValue.INFO_LOGLEVEL);
    public static final TypedValue WARN_SEVERITY = new TypedValue("UCTE_WARN", TypedValue.WARN_LOGLEVEL);
    public static final TypedValue ERROR_SEVERITY = new TypedValue("UCTE_ERROR", TypedValue.ERROR_LOGLEVEL);
}
```

They are used during import:
```java
...
if (line.getResistance() < ZERO_EPS) {
    reporter.report(Report.builder()
        .withKey("negativeLineResistance")
        .withDefaultMessage("${lineId} - Real line resistance cannot be negative (${resistance} ohm)")
        .withValue("lineId", lineId)
        .withTypedValue("resistance", line.getResistance(), TypedValue.RESISTANCE)
        .withSeverity(UcteReportConstants.ERROR_SEVERITY)
        .build());
...
```

* **What is the expected behavior?**
Instead of a different *type* for each severity level, define a type for Severity, then define generic constants for a set of predefined level values.

```java
public class TypedValue {
    // Types
    public static final String UNTYPED = "UNTYPED";
    public static final String SEVERITY = "SEVERITY";
    public static final String ACTIVE_POWER = "ACTIVE_POWER";
    public static final String REACTIVE_POWER = "REACTIVE_POWER";
    ...
    // Typed constants for predefined values
    public static final TypedValue INFO_SEVERITY = new TypedValue("INFO", TypedValue.SEVERITY);
    public static final TypedValue WARN_SEVERITY = new TypedValue("WARN", TypedValue.SEVERITY);
    public static final TypedValue ERROR_SEVERITY = new TypedValue("ERROR", TypedValue.SEVERITY);
    ...
````

These constants will be seen/shared by all uses of the functional logs API. After the change:

The report builder `withSeverity` method could check that the given object value is of the type Severity.
```java
    public ReportBuilder withSeverity(TypedValue severity) {
        if (severity.type != TypedValue.SEVERITY) throw new Exception(...);
        values.put(Report.REPORT_SEVERITY_KEY, severity);
        return this;
    }
```

The class UcteReportContants could be removed, and the UCTE import could be written as:
```java
...
if (line.getResistance() < ZERO_EPS) {
    reporter.report(Report.builder()
        .withKey("negativeLineResistance")
        .withDefaultMessage("${lineId} - Real line resistance cannot be negative (${resistance} ohm)")
        .withValue("lineId", lineId)
        .withTypedValue("resistance", line.getResistance(), TypedValue.RESISTANCE)
        .withSeverity(ReporterConstants.ERROR_SEVERITY)
        .build());
...
```

If a user wants to add a specific severity level, it could just create a new `TypedValue` constant in its own module.

* **What is the motivation / use case for changing the behavior?**
Avoid code duplication and clarify use of severity attribute in functional logs.

* **Please tell us about your environment:**
  - PowSyBl Version: 4.6.0
  - OS Version: ALL

geofjamg, 2022-01-26:
Your proposition looks fine to me.

flo-dup, 2022-01-27:
I agree with having default severities to avoid the duplication, as we often need those 5 classical trace/debug/info/warn/error. And if a reporter user needs a specific severity he can add it on its side. For instance we could think of a specific severity `UcteReportConstants.CRITICAL_SEVERITY`.

But then, thinking on the UI side, @geofjamg how will you display that specific severity, which you might not know of? Meaning, which colour, grey, red? how do you know whether you have to display with filter ">= ERROR"? That was the purpose of having those 5 types for severities, instead of just 1.

flo-dup, 2022-02-11:
After some further thoughts and discussion, I'm ok with your proposition @zamarrenolm.
Then filtering logs on the UI will not be possible with > or expression, but one can imagine a list of checkbox of already known/seen severities, and a grey colour for unknown severities.

PR #1995 (2022-03-07): Refactor support for severity in functional logs

**Please check if the PR fulfills these requirements** *(please use `'[x]'` to check the checkboxes, or submit the PR and then click the checkboxes)*
- [X] The commit message follows our guidelines
- [X] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem ?** *If so, link to this issue using `'#XXX'` and skip the rest*
Fix #1950
closed 2022-03-07 by miovd