powsybl/powsybl-core #1954: Increasing `ShortCircuitAnalysisResult`

annetill, 2022-01-26:
* **Do you want to request a *feature* or report a *bug*?**

A feature. For the moment, the results of a symmetrical short circuit calculation are the limit violations and the fault result. The fault result is a simple class with:
```
public final class FaultResult extends AbstractExtendable<FaultResult> {

    private final String id;

    private final float threePhaseFaultCurrent;
```
For the moment, and as it is written in the java doc, the `id` could be anything but for the private implementation the `id` is the voltage level `id`. But after calculation we get the three phase current by bus (from the bus view indeed) and the part of each equipment connected to the bus of this three phase current. We want to have access to these results. Is it a generic result ?

* **What is the current behavior?**

See above.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **What is the expected behavior?**

For security analysis, we have a class called `BusResults`. We can imagine having increasing `FaultResult` to store the id of the bus and a list or a map to given for each connectable connected to the bus the contribution to the three phase current of the bus. Any idea ? 

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

sylvlecl, 2022-01-28:
I agree it would be better to have the ID of the bus available.

The reason behind using voltage level ID, like in voltage violations, was that calculated buses are not "stable" and therefore the ID of the bus might be invalid shortly after the computation. However I think it's a far bigger problem to not be able at all to differentiate the result of 2 buses in the same voltage level.

annetill, 2023-10-11:
Solved !