powsybl/powsybl-core #1984: LF parameters extensions: loading from map + link with LF provider

sylvlecl, 2022-02-21:
* **Do you want to request a *feature* or report a *bug*?**

Feature

* **What is the current behavior?**

- We have no link, in the API, between a LF provider and its specific parameters
- parameters cannot be loaded from a plain list of properties (key/value pairs)

It makes impossible the creation of LF-specific parameters in a way which would be independent from the implementation.
For example, in `pypowsybl`, we cannot create LF specific parameters from a python dictionary.

* **What is the expected behavior?**

1. we should provide a link between LF provider and their parameters
2. we should provide a method, in LF parameters extensions (and LF parameters?), to load parameters from any list of key/value pair:

* **What is the motivation / use case for changing the behavior?**

See above, being able to write something like this in pypowsybl:
```python
pp.loadflow.run_ac(network, provider='OpenLoadFlow', provider_params={ 'param1': 'value1', ... }
```

sylvlecl, 2022-02-21:
**Design notes:**

In the current state, the features of a LF param extension are dispatched among at least 2 services:

1. The one which allows to load it from config (`ConfigLoader`)
2. The one which allows to serialize/deserialize to/from JSON

I think it would make more sense to centralize more those features into the LF provider.
We could then have something like :
```java
interface LoadFlowProvider {
...
    // Defines the JSON serializer/deserializer to use
    ExtensionSerializer<E extends Extension<LoadFlowParameters>> getParametersExtensionSerializer();
    
    // Load specific parameters from any map
    Extension<LoadFlowParameters> loadSpecificParameters(Map<String, String> properties);

    // Load from platform config : will probably share code with the one above
    Extension<LoadFlowParameters> loadSpecificParameters(PlatformConfig config);
}
```

sylvlecl, 2022-02-21:
**After discussion:**

1. Let's put the JSON serializer in the provider, but with the explicit JSON interface ExtensionJsonSerializer
2. Let's treat that in 2 evolutions:
    - First refactor the provider to include loading from platform config and JSON serializer.
      We need to do it for all computations: loadflow, security analysis, sensitivity analysis
    - Then add to it the loading from map

geofjamg, 2022-02-22:
Maybe also add Optional ?

```
interface LoadFlowProvider {
...
    // Defines the JSON serializer/deserializer to use
    Optional<ExtensionJsonSerializer<E extends Extension<LoadFlowParameters>>> getParametersExtensionSerializer();
    
    // Load specific parameters from any map
    Optional<Extension<LoadFlowParameters>> loadSpecificParameters(Map<String, String> properties);

    // Load from platform config : will probably share code with the one above
    Optional<Extension<LoadFlowParameters>> loadSpecificParameters(PlatformConfig config);
}
```

sylvlecl, 2022-03-17:
Reopening because #1987 only prepared the ground:
now we can add the method to load from a map of properties, in the `LoadFlowProvider`.

miovd, 2022-04-22:
Resolved by #2033

closed 2022-04-22 by miovd