powsybl/powsybl-core #1990: Importers method do not support null object for DataSource, Path, InputStream, etc.

miovd, 2022-03-01:
* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
Null objects in Importers method are processed without any throwed exception.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
This has been discussed [here](https://github.com/powsybl/powsybl-core/pull/1986#discussion_r816259664). There are tests that show this behavior.

* **What is the expected behavior?**
An exception should be thrown: objects should be required not null.

* **What is the motivation / use case for changing the behavior?**
Preventing NPE along the way

samirromdhani, 2026-01-28:
This issue was opened as a result of this  [discussion](https://github.com/powsybl/powsybl-core/pull/1986#discussion_r816259664) in 2022. At that time, a test using a `null` datasource was passing, showing that Importers methods accepted and processed null inputs without throwing any exception.

Since then, the code has changed:

- The original test, class, and package have been removed.
- The related behavior seems to have been moved to` iidm.serde` package, in `XMLImporterTest` class.

When add an equivalent test today (using a null datasource), 

```java
    @Test
    public void importDataTest(){
        Network network = importer.importData((ReadOnlyDataSource) null, ...
        ...
```
The test no longer passes, so the behavior is now different.
I suggest close as completed.

AzazelSensei, 2026-09-02:
Opened https://github.com/powsybl/powsybl-core/pull/4083 — `Importers.importData` now throws on a null DataSource.

samirromdhani, 2026-09-02:
Thanks. The null check now happens directly in Importers