powsybl/powsybl-core #2024: PSSE Transformer conversion

cesar-luis-galli, 2022-03-18:
* **Do you want to request a *feature* or report a *bug*?**
bug

* **What is the current behavior?**
Shunt admittance is not correctly corrected using tap changer admittance correction

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
See attached pdf, which explains in detail the issue.

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)
[trasformer_models.pdf](https://github.com/powsybl/powsybl-core/files/8307120/trasformer_models.pdf)

zamarrenolm, 2022-03-21:
Thank you for comments. We will review in detail.

Have you taken into account that `step.G1` and `step.B1` are percentages? They are not additive quantities to `Ysh`, the actual shunt admittance for a given step is `Ysh,step = Ysh * (1 + step.Y)/100`.

cesar-luis-galli, 2022-03-21:
Thanks for your reply.

Yes I did.
The final comment in the attachment notes that step's variations must be
initialized with Ysh expressed as percent of base admittance.
Propossed correction is applyed over that value, so no other operation is
needed, as I see.

Kind regards.

zamarrenolm, 2022-03-23:
We have reviewed the calculations and we think they are correct.

You have to take into account that `step.G1`, `step.B1`, in PowSyBl, are defined as percentages of the transformer `G,B`. That definition can not be changed, as the rest of the PowSyBl framework assumes that the quantities for the corrections given in each step will modify the transformer `G,B` to obtain the current (effective) `G',B'` in this way:
```
G' = G * (1 + step.G1 / 100)

B' = B * (1 + step.B1 / 100)
```
Of course the corrections stored in `step.G1, step.B1` are constants, they are used as a factor that multiplies the transformer base `G,B` to obtain a corrected `G',B'`.

A clarification document is attached: [issue-2024.pdf](https://github.com/powsybl/powsybl-core/files/8331673/issue-2024.pdf)

cesar-luis-galli, 2022-03-23:
As I see from the document (and after reviewing docs in powsybl.org), my confusion is caused by the term "base".  For me, nominal it's no synonym of base.  I understand "base" as base values used as p.u. basis, so I understood that "base"  values for variations where calculated using rated power and nominal winding voltage (Zbase=Un^2/Srated), which are all constants independent from the actual value of the modeled impedances/admittances (that's why I thought that resulting correction was **absolutely** constant, 'cause the factor was applied over another constant value: either Zbase or Ybase).
After your clarification, I see that step variations are calculated relative to **nominal** impedance/admittance.

Thanks for your attention.
I'll close the issue.

Best regards

closed 2022-03-23 by cesar-luis-galli