powsybl/powsybl-core #2068: CGMES export : potential duplication of Topological Nodes for ends of AC/DC converters

zamarrenolm, 2022-04-08:
* **Do you want to request a *feature* or report a *bug*?**
bug

* **What is the current behavior?**
if two different IIDM DC lines, with its corresponding converters, end in the same AC bus, two topological nodes with the same ID will be exported to the CGMES TP file.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
No duplicated identifiers should be present in the CGMES files.

* **What is the expected behavior?**
Topological Node should be defined only once in the CGMES exported file.

zamarrenolm, 2022-04-08:
The definition of TNs in the substation has to be reviewed:

End buses of converters are exported as Topological Nodes with `ID = busID + "DC"`.

But the IDM AC bus that is the end of the converter will also be exported as a Topological Node with `ID = busID`.

miovd, 2022-04-11:
Resolved by #2063

closed 2022-04-11 by miovd