powsybl/powsybl-core #2072: Extension adder, return type of add method

geofjamg, 2022-04-11:
* **Do you want to request a *feature* or report a *bug*?**
bug

* **What is the current behavior?**
```java
@Override
    public SubstationPosition read(Substation substation, XmlReaderContext context) throws XMLStreamException {
        Coordinate[] coordinate = new Coordinate[1];
        XmlUtil.readUntilEndElement(getExtensionName(), context.getReader(), () -> {
            double longitude = XmlUtil.readDoubleAttribute(context.getReader(), "longitude");
            double latitude = XmlUtil.readDoubleAttribute(context.getReader(), "latitude");
            coordinate[0] = new Coordinate(latitude, longitude);
        });
        substation.newExtension(SubstationPositionAdder.class)
                .withCoordinate(coordinate[0])
                .add();
        return substation.getExtension(SubstationPosition.class);
    }
```

* **What is the expected behavior?**
`substation.newExtension` should return the extension and not the adder to avoid re-getting the extension with `substation.getExtension(SubstationPosition.class)`

* **What is the motivation / use case for changing the behavior?**
API usability

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

geofjamg, 2022-04-12:
I report yesterday discussions:
  - it is not the classical design of builder pattern to return the builder instead of the built object
  - the original reason for doing this seems to be the chaining of multiple extension creation BUT:
    - there is very few need for that (never seen one example of code for this usage)
    - most of the time we want to get the extension it self (see all XML deserializer)

alicecaron, 2026-01-08:
resolved by https://github.com/powsybl/powsybl-core/pull/2246

closed 2026-01-08 by alicecaron