powsybl/powsybl-core #2103: [CGMES] Fill ENTSOE category extension

sylvlecl, 2022-05-05:
* **Do you want to request a *feature* or report a *bug*?**

Feature.

* **What is the current behavior?**

We have a `GeneratorEntsoeCategory` extension, but it's not created when importing a CGMES file.
The information is lost.

* **What is the expected behavior?**

We should retrieve that information and create the extension to keep it.

* **What is the motivation / use case for changing the behavior?**

pypowsybl users expected to have that information filled in, when importing a CGMES file.

@zamarrenolm : would you know how to retrieve that information ?

geofjamg, 2022-05-05:
Convergence get it by parsing the description field of generating units... No idea if it is an official hack agreed by everybody.

zamarrenolm, 2022-05-10:
I planned to write it as a post-processor, instead of adding the "hack" directly to conversion, as it seems it is something not really standardized. Any thoughts on that ?

geofjamg, 2022-05-10:
> I planned to write it as a post-processor, instead of adding the "hack" directly to the conversion code, as it seems it is something not really standardized. Any thoughts on that ?

Good idea!

PR #2112 (2022-05-11): Fill ENTSOE category extension from CGMES generating unit description

**Please check if the PR fulfills these requirements** *(please use `'[x]'` to check the checkboxes, or submit the PR and then click the checkboxes)*
- [X] The commit message follows our guidelines
- [X] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem ?** *If so, link to this issue using `'#XXX'` and skip the rest*
Closes #2103
closed 2022-05-11 by annetill