powsybl/powsybl-core #2171: Implementation of Topological change action

Hadrien-Godard, 2022-06-16:
* **Do you want to request a *feature* or report a *bug*?**
Feature.

* **What is the current behavior?**
The action is not implemented.

* **What is the expected behavior?**
The Topological change action is implemented as an implementation of the Action class (see issue #2015).
This action consists in opening or closing a network element.
A starting idea of possible implementation is:
```java
class TopologicalAction extends Action {
    String elementId;
    boolean open;
}
```

* **What is the motivation / use case for changing the behavior?**
Allow security analysis to take this action type into account.

annetill, 2023-03-02:
We have a switch action that can be used for node/breaker models. But it means that a lot of switches will be kept for the security analysis with remedial actions. That is why we also have a line connection action. Do we need something similar for transformers (two and three windings) and for injections?

annetill, 2024-07-01:
We have now `TerminalsConnectionAction`, which for me solved this issue. @Hadrien-Godard can we close it?

closed 2024-07-01 by annetill

Hadrien-Godard, 2024-07-01:
> We have now `TerminalsConnectionAction`, which for me solved this issue. @Hadrien-Godard can we close it?

Yes