powsybl/powsybl-core #2176: Implementation of TCSC compensation value action

Hadrien-Godard, 2022-06-16:
* **Do you want to request a *feature* or report a *bug*?**
Feature.

* **What is the current behavior?**
The action is not implemented.

* **What is the expected behavior?**
The TCSC compensation value action is implemented as an implementation of the Action class (see issue #2015).
This action consists in changing the compensation value of a TCSC.
A starting idea of possible implementation is:
```java
class TcscCompensationAction extends Action {
    String tcscId;
    double compensationValue;
}
```

* **What is the motivation / use case for changing the behavior?**
Allow security analysis to take this action type into account.

sylvlecl, 2022-06-17:
What is TCSC? :)
Maybe we should make the name more explicit (or javadoc may be enough).

Hadrien-Godard, 2022-06-22:
Sorry :)
It is the acronym for Thyristor Controlled Series Compensator. It is a kind of FACTS (Flexible Alternating Current Transmission System) that modifies line impedance to control power flowing through a line.

annetill, 2023-03-02:
Is it more complex that just having a remedial action with the id of the action (of course), the id of the line on which the TCSC is installed and a delta of impedance (maybe only `+/-X` here or it could be `+/-R` too?) ? The action will just applied to the line during calculation line.getX() becomes line.getX() + action.getDeltaX() ? It is possible to have that can of device somewhere else ? The generic name of this action could be `LineImpedanceAction` (or line reactance action).