powsybl/powsybl-core #2183: Security analysis lost load/generation

geofjamg, 2022-06-28:
* **Do you want to request a *feature* or report a *bug*?**
Feature

* **What is the current behavior?**
When running a security analysis, for some contingencies, we could endup with a loss of generation or loss.
This is not reported in security analysis result API.

* **What is the expected behavior?**
We should add to PostContingencyResult, the amount of generation and load lost because of the contingency.

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

sylvlecl, 2022-06-28:
Note that it will be interesting also for the consequences of actions.
In internal project, we not only display a global amount, but a more detailed list of loads and generators.

geofjamg, 2022-06-28:
> In internal project, we not only display a global amount, but a more detailed list of loads and generators.
Yes, right we should be able to store the list of lost injections and associated MW