powsybl/powsybl-core #2193: Reporter variable context inheritance

geofjamg, 2022-07-03:
* **Do you want to request a *feature* or report a *bug*?**
Feature

* **What is the current behavior?**
When formatting reports, we should be able to use
 - variables of the report
 - variable of all parent reporters

!!!! In the actual text formatter, we only use variable of the report and the ones of first parent reporter.

This would be a huge improvement in term of compactness of report as it would be possible to store common variables in a common reporter ancestor and no need to repeat it in all child reporters.

* **What is the expected behavior?**
The evolution is
 1) mostly a matter of documentation of the API to describe that it is allowed to define variables in any parent of a reporter and that all formatters should take them into account.
 2) implement it in default text formatter as a demonstrator.

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

flo-dup, 2022-07-06:
Following a discussion with @laurgarn I'm wondering if we shouldn't also do the opposite, at least for a depth of one. Why? because the title cannot contain results which are in its child messages, and even if this node is not expanded we might want to display some concise info about what happened within.

laurgarn, 2022-07-07:
@flo-dup aggregation/synthetis might be simpler/more flexible on the front-end (that's what is done in Convergence, where transmitted contexts don't have a status level by themselves).
Of course back-end aggregation might be more powerful but yet I can only think of worst child status as a standard aggregation rule. May be overall processing time, but this is not as easy as in C++.
On the other side, front-end could have parameters to aggregate differently the same back-end provided data.

PR #2201 (2024-03-11): Reporter new API
**Please check if the PR fulfills these requirements**
- [x] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem ?** 
Yes, fixes #2190, fixes #2191, fixes #2193, deprecates #2192, deprecates #2192

**Does this PR introduce a breaking change or deprecate an API?**
- [x] Yes
- [ ] No

**If yes, please check if the following requirements are fulfilled**
- [x] The *Breaking Change* or *Deprecated*  [... cut]
closed 2024-03-11 by flo-dup