powsybl/powsybl-core #2194: Reporter: provide a TypedValue constructor overload for undefined type

geofjamg, 2022-07-03:
* **Do you want to request a *feature* or report a *bug*?**
Feature

* **What is the current behavior?**
We should provide a `TypedValue` constructor overload in case of an undefined type.

```java
reporter.createSubReporter("lfNetwork", "Network CC${networkNumCc} SC${networkNumSc}",
                Map.of(NETWORK_NUM_CC, new TypedValue(networkNumCc, TypedValue.UNTYPED),
                        NETWORK_NUM_SC, new TypedValue(networkNumSc, TypedValue.UNTYPED)));
```

* **What is the expected behavior?**

could be written 
```java
reporter.createSubReporter("lfNetwork", "Network CC${networkNumCc} SC${networkNumSc}",
                Map.of(NETWORK_NUM_CC, new TypedValue(networkNumCc),
                        NETWORK_NUM_SC, new TypedValue(networkNumSc)));
```

Or even better
```java
reporter.createSubReporter("lfNetwork", "Network CC${networkNumCc} SC${networkNumSc}",
                Map.of(NETWORK_NUM_CC, TypedValue.of(networkNumCc),
                        NETWORK_NUM_SC, TypedValue.of(networkNumSc)));
```

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

sylvlecl, 2022-07-04:
Maye `TypedValue.untyped(value)` to be more explicit ?

flo-dup, 2024-03-11:
TypedValue cannot be instantiated anymore, adder/builder should be used instead with methods:
- `withTypedValue(valueKey, value, type)`
- `withUntypedValue(valueKey, value)`

closed 2024-03-11 by flo-dup