powsybl/powsybl-core #2211: [convert-network] input-file mismatch

obretteville, 2022-07-12:
* **Do you want to request a *feature* or report a *bug*?**
bug

* **What is the current behavior?**
for command (conversion from UCT format to XIIDM format):
    itools convert-network --input-file elementName.uct --output-file elementName.xiidm --output-format XIIDM
if elementName.mat exists, it is used for conversion although elementName.uct is specified as input file 

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
itools convert-network --input-file elementName.uct --output-file elementName.xiidm --output-format MATPOWER
itools convert-network --input-file elementName.uct --output-file elementName.xiidm --output-format XIIDM
The first command output elementName.mat (although elementName.xiidm is specified as output file)
The second command uses elementName.mat as input file (although elementName.xiidm is specified as input file)

* **What is the expected behavior?**
The first command should output elementName.xiidm (with a matpower content), not elementName.mat
The second command should use given name as input-file (elementName.uct), not a different file (elementName.mat if this file exists).

* **What is the motivation / use case for changing the behavior?**
Command must use given parameters.

* **Please tell us about your environment:**
  - PowSyBl Version:4.9
  - OS Version:Ubuntu 20.04

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

geofjamg, 2022-07-25:
This is not specific to command line execution but a well known issue on powsybl [data sources](https://github.com/powsybl/powsybl-core/tree/main/commons/src/main/java/com/powsybl/commons/datasource).

geofjamg, 2022-07-25:
Here is a test case to reproduce the bug

olperr1, 2024-08-27:
Solved by #3102

closed 2024-08-27 by olperr1