powsybl/powsybl-core #2215: CGMES assembled model support

geofjamg, 2022-07-14:
* **Do you want to request a *feature* or report a *bug*?**
feature

* **What is the current behavior?**
We need to provide a zip file with all profiles, to import a CGMES.

* **What is the expected behavior?**
We should also support a single file loading containing the assembled model.

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

zamarrenolm, 2022-07-15:
I think the main issues are related to boundary management, the rest of the data could come in a single file without problems. 

If we let the boundary files also be assembled with the rest of the network model we have to make a small change to identify which are the nodes at the boundaries. Currently, all the nodes (connectivity or topological) defined in an instance file that has a "boundary" profile are labelled as belonging to boundary and thus receiving a special treatment during import.

An assembled model in a single instance file may contain data from multiple profiles:

```xml
<rdf:RDF  xmlns:cim="http://iec.ch/TC57/2013/CIM-schema-cim16#" xmlns:entsoe="http://entsoe.eu/CIM/SchemaExtension/3/1#" xmlns:md="http://iec.ch/TC57/61970-552/ModelDescription/1#" xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#">
    <md:FullModel rdf:about="urn:uuid:146bd36c-93c4-9848-baaf-761a91002420">
        <md:Model.created>2022-07-01T17:10:22Z</md:Model.created>
        <md:Model.description>Assembled by ...</md:Model.description>
        <md:Model.version>1</md:Model.version>
        <md:Model.scenarioTime>2021-03-10T09:30:00Z</md:Model.scenarioTime>
        <md:Model.modelingAuthoritySet>http://tscnet.eu/EMF</md:Model.modelingAuthoritySet>
    <md:Model.modelingAuthoritySet>http://www.tso-example.com</md:Model.modelingAuthoritySet>
        <md:Model.profile>http://entsoe.eu/CIM/EquipmentBoundary/3/1</md:Model.profile>
        <md:Model.profile>http://entsoe.eu/CIM/EquipmentBoundaryOperation/3/1</md:Model.profile>
        <md:Model.profile>http://entsoe.eu/CIM/TopologyBoundary/3/1</md:Model.profile>
        <md:Model.profile>http://entsoe.eu/CIM/EquipmentCore/3/1</md:Model.profile>
        <md:Model.profile>http://entsoe.eu/CIM/EquipmentOperation/3/1</md:Model.profile>
        <md:Model.profile>http://entsoe.eu/CIM/SteadyStateHypothesis/1/1</md:Model.profile>
        <md:Model.profile>http://entsoe.eu/CIM/Topology/4/1</md:Model.profile>
        <md:Model.profile>http://entsoe.eu/CIM/StateVariables/4/1</md:Model.profile>
    </md:FullModel>
```

A way to identify all boundary nodes would be to assume all of them must belong to a specific geographical or sub-geographical region. In the case of Europe transmission boundaries, there is a region named "ENTSO-E":

[image]

Haigutus, 2022-07-15:
Boundary points are identifiable without assuming that they are all in one file, there is a flag to identify it in CGMES 2.4.

[image]

CGMES 3.0 has a different logic, boundary point is in its own class, referring to CN

[image]

zamarrenolm, 2022-07-18:
Updated #2216 to use `boundaryPoint` attribute in CGMES 2.4

PR #2216 (2022-08-18): Allow boundaries to be assembled with other network data in the same instance file

**Please check if the PR fulfills these requirements** *(please use `'[x]'` to check the checkboxes, or submit the PR and then click the checkboxes)*
- [X] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem ?** *If so, link to this issue using `'#XXX'` and skip the rest*
Closes #2215.
closed 2022-08-18 by miovd