powsybl/powsybl-core #2229: Renaming AttachNewLineOnLine / AttachVoltageLevelOnLine

flo-dup, 2022-08-04:
**Do you want to request a *feature* or report a *bug*?**
None of them, renaming

**What is the current behavior?**
- `AttachVoltageLevelOnline` to insert (/attach/connect) an existing voltage level on a line by creating a [line tap (or tee point)](https://www.electropedia.org/iev/iev.nsf/display?openform&ievref=601-02-29) on that line, therefore cutting the existing line in two.
- `AttachNewLineOnLine` to insert (/attach/connect) similarly an existing voltage level on a line by creating a fictitious [switching voltage level](https://www.electropedia.org/iev/iev.nsf/display?openform&ievref=605-01-02) on that line, therefore also cutting the existing line in two.

**What is the expected behavior?**
The "Attach" term does not seem to be used a lot in that context. "Connect" or "Insert" seem more appropriate.

Besides, the "line tap" or "tee point" (*point de piquage* in french) and the "switching substation" (*poste de coupure* in french) terms which are referenced by IEC are not mentioned.

**What is the motivation / use case for changing the behavior?**
Having a more understandable name. 

Nonetheless, for its cons, the current javadoc is pretty clear, and the term attach even if not common is still understandable. Is it worth the breaking change?

**Other information**
IEC links:
line tap: https://www.electropedia.org/iev/iev.nsf/display?openform&ievref=601-02-29
tapped substation: https://www.electropedia.org/iev/iev.nsf/display?openform&ievref=601-02-21
switching substation: https://www.electropedia.org/iev/iev.nsf/display?openform&ievref=605-01-02

geofjamg, 2022-08-04:
Yes it worth a breaking change. It is a quite new feature, so nobody worried about the change.

annetill, 2022-08-16:
I understand. The renaming could help understanding better this feature but we should also rely on documentation. These two methods are already well documented because they cannot be used without any documentation, and we can imagine to add the terms "line tap" or "tee point" and "switching substation", and the links to IEC. 

The method `AttachVoltageLevelOnline` could be renamed, but what do you suggest? I understand that you want to replace attach by connect or insert. But adding switching substation is confusing because we create a voltage level that does not belong to any substation but to the network. In the method it is really the voltage level that is inserted. So it could be `InsertSwitchingVoltageLevelOnLine` or `ConnectSwitchingVoltageLevelOnLine`. I think I prefer the last one.

The method  `AttachNewLineOnLine`, we create a tee point yes, but we also connect a new line to this point. It is a bit different that above. So, if you really want to have this "tee point term", we could imagine something like `ConnectNewLineOnLineThroughTeePoint`?

PR #2238 (2022-08-31): Rename topology network modifications

**Please check if the PR fulfills these requirements** *(please use `'[x]'` to check the checkboxes, or submit the PR and then click the checkboxes)*
- [x] The commit message follows our guidelines
- [x] Tests for the changes have been added (for bug fixes / features)

**Does this PR already have an issue describing the problem ?** *If so, link to this issue using `'#XXX'` and skip the rest*
Fix #2229 

**What kind of change does this PR introduce?** *(Bug fix, feature, docs update, ...)*
Renaming

**Does this PR introduce a breaking change or deprecate a [... cut]
closed 2022-08-31 by miovd