powsybl/powsybl-core #2234: ACDCTerminal.sequenceNumber is optional

geofjamg, 2022-08-09:
* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
CGMES import requires ACDCTerminal.sequenceNumber to be set

* **What is the expected behavior?**
According to specification, this attribute is optional, and CGMES importer should work without it.

* **What is the motivation / use case for changing the behavior?**
To be compliant with CGMES standard

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

zamarrenolm, 2022-08-17:
To establish terminals 1 and 2 of each branch we will use `sequenceNumber` if available. If `sequenceNumber` is not present, we will sort the terminals by its identifier. 

An observation about SPARQL related queries: we have to be careful when adding the `FILTER` condition to select the proper order of terminals. For AC Line Segments, we have checked that if the `OPTIONAL` for the `gch` attribute is placed **before** the `FILTER` clause, the query takes too long even for small networks. The difference of execution time between these two queries is very large (tested with RDF4J, for now):

Select all ACLSs with terminals ordered by identifier, takes less than 100ms even for big networks (more than 2000 ACLSs):
```sparql
SELECT *
WHERE {
{ GRAPH ?graph {
    ?ACLineSegment
        a cim:ACLineSegment ;
        cim:ACLineSegment.r ?r ;
        cim:ACLineSegment.x ?x ;
        cim:ACLineSegment.bch ?bch ;
        cim:IdentifiedObject.name ?name .
    ?Terminal1
        a cim:Terminal ;
        cim:Terminal.ConductingEquipment ?ACLineSegment .
    ?Terminal2
        a cim:Terminal ;
        cim:Terminal.ConductingEquipment ?ACLineSegment .
    FILTER ( str(?Terminal1) < str(?Terminal2) )
    OPTIONAL {
        ?ACLineSegment cim:ACLineSegment.gch ?gch
    }
    OPTIONAL {
        ?ACLineSegment cim:Equipment.EquipmentContainer ?Line .
        ?Line
            a cim:Line ;
            cim:IdentifiedObject.name ?lineName
    }
}}
}
```

Select all ACLSs first with its optional attributes, then terminals ordered by identifier, takes more than 5s for small networks (less than 500 ACLSs):
```sparql
SELECT *
WHERE {
{ GRAPH ?graph {
    ?ACLineSegment 
        a cim:ACLineSegment ;
        cim:ACLineSegment.r ?r ;
        cim:ACLineSegment.x ?x ;
        cim:ACLineSegment.bch ?bch ;
        cim:IdentifiedObject.name ?name .
    OPTIONAL {
        ?ACLineSegment cim:ACLineSegment.gch ?gch
    }
    ?Terminal1
        a cim:Terminal ;
        cim:Terminal.ConductingEquipment ?ACLineSegment .
    ?Terminal2
        a cim:Terminal ;
        cim:Terminal.ConductingEquipment ?ACLineSegment .
    FILTER ( str(?Terminal1) < str(?Terminal2) )
    OPTIONAL {
        ?ACLineSegment cim:Equipment.EquipmentContainer ?Line .
        ?Line
            a cim:Line ;
            cim:IdentifiedObject.name ?lineName
    }
}}
}
```

geofjamg, 2022-08-17:
Interesting, maybe this is something to report to rdf4j project?

zamarrenolm, 2022-08-17:
I will dig a little bit more on the issue. The query explanations show the issue, but it is hard to find why the plans are different. I will try it with latest version of rdf4j (4.1.0, we are using 4.0.3): it contains "major performance improvements".

BTW, the Jena version we were using before (3.0.0) performs equally bad with both alternatives. For small-medium networks (more than 500 ACLSs), both alternatives take more than 10s. Also will try to check with more recent version of the library.

closed 2022-08-23 by miovd