powsybl/powsybl-core #2373: Remove limit violation filter from security analysis SPI

sylvlecl, 2022-11-17:
* **Do you want to request a *feature* or report a *bug*?**

Removal of obsolete component.

* **What is the current behavior?**

The security analysis providers are expected to take into account a `LimitViolationFilter` which can remove some violations from the result.

I would propose to remove that parameter from the SPI because:

- its behaviour can, and probably should, be implemented as post processing of the result
- it's redundant with the violation detector (which design and presence in the SPI is also questionable, indeed)

**This is a proposition, to be discussed.**

geofjamg, 2022-11-17:
Ok, I would also remove the violation detector as it is impl related

alicecaron, 2026-08-17:
The  `LimitViolationDetector` has been removed from the `SecurityAnalysis` since this issue: https://github.com/powsybl/powsybl-core/pull/2943