powsybl/powsybl-core #2387: Bus-branch export produces an invalid XIIDM file

sylvlecl, 2022-12-02:
* **Do you want to request a *feature* or report a *bug*?**

Bug

* **What is the current behavior?**

See unit test in https://github.com/powsybl/powsybl-core/pull/2386

When exporting to an XIIDM file in bus-branch topology, elements may have no associated connectable bus.
It causes the XIIDM file to be invalid (impossible to import).

* **What is the expected behavior?**

To be decided:
- do not export those elements ? but it can be difficult to handle if it is referenced elsewhere
- try harder to find a connectable bus ?

sylvlecl, 2022-12-05:
@annetill @miovd @geofjamg @EtienneLt 
After discussion:
it appears it is "known" that the bus-branch export is not supported

--> we should remove completely the support for that export parameter

Is it agreed ?
Of course it's a breaking change. We can document that it could be replaced by the bus-breaker export where the retained switches are configured according to the needs.

miovd, 2022-12-05:
> @annetill @miovd @geofjamg @EtienneLt After discussion: it appears it is "known" that the bus-branch export is not supported
>
> --> we should remove completely the support for that export parameter [...]

Please note before taking a decision that there are users of the feature (dynawo project and ELM for example use the bus-branch export to run their calculations).