powsybl/powsybl-core #2390: [IIDM] Default values for network element adders

sylvlecl, 2022-12-06:
* **Do you want to request a *feature* or report a *bug*?**

Feature

* **What is the current behavior?**

When creating networks programmatically, almost all attributes of all elements must be explicitly defined.

This makes the creation of simple networks very verbose and fastidious.

* **What is the expected behavior?**

Instead, we could make some attributes optional, and assign them default values.

A first pull request was drafted here: #2003 

Here is a tentative list of default values :

- Line/tie lines/dangling lines/two windings transformers and three windings transformers legs:
  - g/b/g1/b1/g2/b2 = 0
  - r = 0 ?
- Two windings transformers
- Ratio tap changers:
  - r, x, g, b = 0 for every step
- Phase tap changers 
  - r, x, g, b = 0 for every step
  - ratio = 1 for every step
- Shunt compensators:
  - g = 0 for each section

geofjamg, 2022-12-07:
Agree on all your proposition.
For r=0 with line, I am also not 100% convinced, let's keep it mandatory for a first version?
@annetill if you also agree on this list, we can use this spec to terminate your PR?

annetill, 2022-12-08:
For ratio tap changers, an alpha of 0 for every step?

closed 2023-01-25 by EtienneLt