powsybl/powsybl-core #2483: New limit type: `VOLTAGE_ANGLE` or just `ANGLE`

annetill, 2023-03-02:
* **Do you want to request a *feature* or report a *bug*?**

A new feature. 

* **What is the current behavior?**

Nothing exists.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **What is the expected behavior?**

New voltage angle operational limit. Its violation should be reported during a security analysis. As first simple approach, this new kind of limit could be associated to a `Line` or a `TwoWindingsTransformer` with a simple `double` as maximum absolute admissible value for the voltage angle difference between the two buses on which the branch is connected. A second approach, more generic could be to associate this new limit to two different buses of the network. As buses could be in different voltage levels, it means that it will be a limit of `Network` (I don't know if we can do that). As each voltage level can have a node/breaker topology or a bus/breaker topology, the id of the bus could be a bus bar section id or a configured bus id.

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

annetill, 2023-03-02:
https://eepublicdownloads.entsoe.eu/clean-documents/CIM_documents/Grid_Model_CIM/Voltage_Angle_Limit_Profile_Specification_v1.0.pdf

EtienneLt, 2023-03-02:
should the voltage angle difference better be in radian ?

annetill, 2023-03-02:
> should the voltage angle difference better be in radian ?

In our `Network`, angles are in degree, even if calculations used radians. So it is better to remain coherent as the limit will be serializable.

jeandemanged, 2023-03-02:
latest approved CSA 2.1 profile: https://eepublicdownloads.entsoe.eu/clean-documents/CIM_documents/Grid_Model_CIM/EquipmentReliability_Profile_Specification_v2.1.pdf

also be aware that CSA2.2 is ongoing

I might disaggree on putting angle limits on branches, this is a specific case. The general case places limits between any two terminals.

annetill, 2023-03-02:
> latest approved CSA 2.1 profile (now this info is in EquipmentReliability profile): https://eepublicdownloads.entsoe.eu/clean-documents/CIM_documents/Grid_Model_CIM/EquipmentReliability_Profile_Specification_v2.1.pdf
> 
> also be aware that CSA2.2 is ongoing
> 
> I might disaggree on putting angle limits on branches, this is a specific case. The general case places limits between any two terminals (and not necessarily between busbar sections terminals, although this might cover 99% use cases).

I understand the limitation. Maybe the most generic way to model that is to use terminals:
- If we want to monitor a dedicated line, we have to put its terminals ; but it will be associated to a network, which is less efficient that attaching directly the limit to the line. But more generic for sure...  
- If we want to monitor to buses (from the bus/view), we have to follow the logic of `slackTerminal` extension. We give two terminals (reference to an equipment + side if needed) and the limit will be attached to a network. Note that in bus/branch model from CGMES, the id of a bus bar section is an issue: in that case, we have to find an equivalent terminal connected to the same configured bus.

geofjamg, 2023-03-02:
I don't think we should copy what has been done for slack terminal, it is quite complex to cover very rare CGMES cases.
A busbar section id for a node/breaker voltage level and a configured bus id for a bus/breaker voltage level will probably do the job for 99.999% of cases. Or maybe I missed something?
So we would only define the angle to monitor by a pair of ID (busbar section of configured bus).

geofjamg, 2023-03-31:
### Proposition to be able to monitor any of the angle different between 2 arbitrary point of the network:

Add a new `TopologyPoint` feature in IIDM. A topology point is able to point any vertex of a topology graph whatever we are in a node/breaker or a bus/breaker model.
It would look something like:

```java
class TopologyPoint {
    String voltageLevelId;
    int node;
    int busId; 
}
```
For a node/breaker voltage level, we use voltageLevelId + node and for a bus/breaker voltage level we use voltageLevelId + busId (as an implementation detailed we might use inheritance to model this).
Unlike existing `TerminalRef` feature, we are able with a `TopologyPoint` to point graph points that are not connected to a terminal, so this is a more generic concept.
Unlike  `TerminalRef` , we are not able, in addition to a topology location, to describe a direction (like it is used for generator reactive power regulation).  `TerminalRef` could be a directed link if used with a branch. `TopologyPoint` is undirected.

Some utility methods could be added to ease its creation for simple cases like:
```java
var tp = TopologyPoint.fromTerminal(...)
Optional<Terminal> t = tp.resolveToTerminal()
```
Once this feature has been implemented, we are able to create `VoltageAngleLimit`
```java
var n = Network.read(...);
n.newVoltageAngleLimit()
    .from(TopologyPoint.create("vlnb", 0))
    .to(TopologyPoint.create("vlbb", "b1"))
    .withLimit(10) // in °
    .add();

List<VoltageAngleLimit> limits = n.getVoltageAngleLimits();
```

geofjamg, 2023-06-06:
So something like:
```java
var n = Network.read(...);
n.newVoltageAngleLimit()
    .from(TerminalRef.from("gen"))
    .to(TerminalRef.from("line", Branch.Side.ONE))
    .withLimit(10) // in °
    .add();

List<VoltageAngleLimit> limits = n.getVoltageAngleLimits();
```

marqueslanauja, 2023-06-08:
Entsoe has published and approved version 2.2 of the EquipmentReliability_Profile_Specification: (https://www.entsoe.eu/Documents/CIM_documents/Grid_Model_CIM/EquipmentReliability_Profile_Specification_v2.2.pdf) In this version, the `angle` attribute must be positive (greater than zero) and an additional Boolean attribute has been added to indicate the flow direction. We are implementing the code according to this version.

geofjamg, 2023-06-08:
> Entsoe has published and approved version 2.2 of the EquipmentReliability_Profile_Specification: [...]

I do not see any reason to use an additional boolean to model exactly the same thin than a signed angle

marqueslanauja, 2023-06-09:
> > Entsoe has published and approved version 2.2 of the EquipmentReliability_Profile_Specification: (https://www.entsoe.eu/Documents/CIM_documents/Grid_Model_CIM/EquipmentReliability_Profile_Specification_v2.2.pdf) In this version, the `angle` attribute must be positive (greater than zero) and an additional Boolean attribute has been added to indicate the flow direction. We are implementing the code according to this version.
> 
> I do not see any reason to use an additional boolean to model exactly the same than than a signed angle

The `isFlowToRefTerminal`  Boolean attribute allows to consider an additional case. The three considered cases are:

- When the flag is true, the flow is from the operating limit terminal to the reference terminal.
- When the flag is false, the flow is in the other direction.
- When the flag is not defined, both directions are considered.

geofjamg, 2023-06-09:
> The `isFlowToRefTerminal` Boolean attribute allows to consider an additional case. The three considered cases are:
>
>     * When the flag is true, the flow is from the operating limit terminal to the [...]

Ok, I understand

annetill, 2023-10-11:
Solved by #2613 and #2657

closed 2023-10-11 by annetill