powsybl/powsybl-core #2496: Sensitivity analysis theshold

geofjamg, 2023-03-10:
* **Do you want to request a *feature* or report a *bug*?**
Feature

* **What is the current behavior?**
We cannot filtervery low value sensitivity.

* **What is the expected behavior?**
We should be able to define in `SensitivityAnalysisParameter` 4 thresholds (double values):
  - powerPowerSensiValueThreshold
  - voltageVoltageSensiValueThreshold
  - powerVoltageSensiValueThreshold
  - anglePowerSensiValueThreshold

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

geofjamg, 2023-03-10:
@annetill @Hadrien-Godard @jeandemanged is it ok for you?

Hadrien-Godard, 2023-03-10:
> @annetill @Hadrien-Godard @jeandemanged is it ok for you?

Yes

obrix, 2023-03-13:
@geofjamg @Hadrien-Godard What should be the default values for those parameters ? 0.0 for no filtering ?

geofjamg, 2023-03-13:
I would say 0 which would mean not filtering

closed 2023-03-16 by geofjamg