powsybl/powsybl-core #2523: Battery missing in exported CGMES and SVG files

swinkelp, 2023-03-29:
* **Do you want to request a *feature* or report a *bug*?**
bug

* **What is the current behavior?**
When creating a battery with `vl2.newBattery()` in my network, and then generate a CGMES zip file of the network, there is no reference to this battery item in both the exported CGMES _EQ file nor the SVG Single Line Diagram.

No errors or warnings are generated during the compilation/run.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

Step 1: add a Battery to our network's voltage level 2:
```
        vl2.newBattery()
                .setId("Battery")
                .setBus("B1")
                .setMinP(0)
                .setMaxP(80)
                .setTargetP(30)
                .setTargetQ(30)
                .add();
```

Step 2: Add code to generate CGMES zipfile and SVG diagram
```
network.write("CGMES", null, Paths.get(tempDir, "substation_cgmes.zip"));
SingleLineDiagram.draw(network, "S1", Paths.get(tempDir, "substation.svg").toString());
```

Step 3: Run the code
`mvn compile exec:java`

* **What is the expected behavior?**

We expect to have a representation of the Battery in both the CGMES _EQ.xml file and the SVG file, which is not the case currently.

* **Please tell us about your environment:**
  - PowSyBl Version: 2023.0.0
  - OS Version: 
     NAME="Ubuntu"
     VERSION="18.04.6 LTS (Bionic Beaver)

geofjamg, 2023-03-30:
@swinkelp thanks for reporting this issue. 
for SLD it will be fixed in next release. 
for CGMES export, we have to think about what could be the best fit in CGMES model as there is no battery modelling.

annetill, 2023-06-19:
CGMES export done in #2527