powsybl/powsybl-core #2526: [Load flow] Add warning when connecting voltage levels with different nominal voltages

FredericSabot, 2023-03-30:
* **Do you want to request a *feature* or report a *bug*?**

Feature

* **What is the current behavior?**

Currently, it is possible to accidentally connect, say a 400kV bus to a 220kV one which can lead to weird load flow results (in my case, I also obtain different results when using itools and pypowsybl, maybe because of different versions/default parameters).

* **What is the expected behavior?**

Output a warning when computing a load flow for such invalid cases (I guess no warning should be given for connections between 380 and 400KV buses).

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: Ubuntu 22.04

geofjamg, 2023-03-30:
Do you have a small example with such a configuration and given / expected results?

FredericSabot, 2023-03-30:
Upon further examination and installation of a more recent itools distribution, both tools give the expected results (results are actually independent of the nominal_v of buses, and only depend on the generator/transformer ratings. This was not the case with v1.4 of powsybl.

My point was mainly that a warning could be output when a line is used to connect voltage levels with significantly different nominal_v as this is most likely an error from the user.

[image]
[image]
[eurostag_itools_old.txt](https://github.com/powsybl/powsybl-core/files/11113159/eurostag_itools.txt)
[eurostag_pypowsybl.txt](https://github.com/powsybl/powsybl-core/files/11113160/eurostag_pypowsybl.txt)

FredericSabot, 2026-02-27:
If I remember well,
- eurostag_itools_old was generated with an old version of itools (probably with OpenLoadFlow as I did not change the defaults)
- eurostag_pypowsybl.txt was generated with pypowsybl (with whatever was the current version at the time and also probably with OpenLoadFlow).

There are 2 issues:
1. The voltage at V-Test is incorrectly computed to 197kV instead of 147 with the old itools version. But it was fixed with a more recent version (no iidm provided with that one) -> OK
2. Powsybl silently loads an "invalid" iidm. I think the warning should be when loading since it could be problematic for other uses, but I don't have a strong opinion on this.

Note a 380 to 400kV line might be acceptable (?)