powsybl/powsybl-core #2547: PSS/E issue with low impedance line conversion

geofjamg, 2023-04-13:
### Describe the current behavior

When converting such a non transformer branch record:
```
    1111,    2222,'1', 0.00000E+0, 3.26000E-4,   0.00000,    0.00,    0.00,    0.00,  0.00000,  0.00000,  0.00000,  0.00000,1,1,   0.00,  20,1.0000,,,,,,,,/* [                                ] */
```

we get in IIDM:

```xml
    <iidm:line id="..." r="0.0" x="4.890000000000001E-4" g1="0.0" b1="-85.207907293797" g2="0.0" b2="81.79959100204505" .... />
```

So the calculated susceptance has an incorrect value that does not allow a LF to converge.

zamarrenolm, 2023-04-14:
Could you check the definition of the corresponding buses? (voltages, base voltages, ...). Maybe they have different base voltages? If would be useful if you can copy the complete line where they are defined.

geofjamg, 2023-04-14:
> Could you check the definition of the corresponding buses? (voltages, base voltages, ...). Maybe they have different base voltages? If would be useful if you can copy the complete line where they are defined.

Yes there is not same nominal voltages.
Here is a test case:
```java
double sb = 100;
double nominalV1 = 12.5;
double nominalV2 = 12;
double r = 0;
double x = 3.26000E-4;
double gi = 0;
double bi = 0;
double gj = 0;
double bj = 0;
double b = 0;
double rEu = impedanceToEngineeringUnitsForLinesWithDifferentNominalVoltageAtEnds(r, nominalV1, nominalV2, sb);
double xEu = impedanceToEngineeringUnitsForLinesWithDifferentNominalVoltageAtEnds(x, nominalV1, nominalV2, sb);
Complex yEu = new Complex(rEu, xEu).reciprocal();
double g1Eu = admittanceEnd1ToEngineeringUnitsForLinesWithDifferentNominalVoltageAtEnds(yEu.getReal(), gi, nominalV1, nominalV2, sb);
double b1Eu = admittanceEnd1ToEngineeringUnitsForLinesWithDifferentNominalVoltageAtEnds(yEu.getImaginary(), b * 0.5 + bi, nominalV1, nominalV2, sb);
double g2Eu = admittanceEnd2ToEngineeringUnitsForLinesWithDifferentNominalVoltageAtEnds(yEu.getReal(), gj, nominalV1, nominalV2, sb);
double b2Eu = admittanceEnd2ToEngineeringUnitsForLinesWithDifferentNominalVoltageAtEnds(yEu.getImaginary(), b * 0.5 + bj, nominalV1, nominalV2, sb);
System.out.println(b1Eu);
System.out.println(b2Eu);
```

marqueslanauja, 2023-06-28:
The calculated susceptance seems abnormally high, but it is correct. These high values for the susceptance may appear when the parameters of the line are expressed in engineering units and the ends of the line have different nominal voltage. In this case 12 KV and 12.5 KV. The line per-unit reactance (0.000326) creates high shunt susceptance values at both ends of the line when the parameters are expressed in engineering units. The obtained IIDM model is equivalent to the initial PSSE model, with the only difference that it is expressed in engineering units. 
Before running the open loadflow the model is again expressed in per unit quantities, resulting in the same values of the initial PSSE model: in the open loadflow network, the shunt susceptance at both ends of the line are zero.

closed 2026-10-15 by rcourtier