powsybl/powsybl-core #2643: New short circuit parameters

annetill, 2023-07-10:
### Describe the current behavior

The parameters are a bit poor. 

### Describe the expected behavior

New parameters could be added:
- `withLoads`: a boolean. If set to `true` means that loads are taken into account in the short-circuit calculation. If set to `false`, all loads are considered with `P0 and `Q0` equal to zero. 
- `withShuntCompensators`
- `withVSCconverterStations`
- `withNeutralPositon` for tap changers only. If `false`, the step defined in the network model is used, else the neutral step is used (rho = 1, alpha = 0).
-  Initial voltage profile: using nominal voltages or configured with a list of <Nominal voltage value, coefficient>.

colinepiloquet, 2023-07-31:
For the shunts, I think that if the boolean is true, a load is created linked to the shunt. And that B of the shunt is used for the load characteristics.

colinepiloquet, 2023-09-18:
Fixed in #2638

closed 2023-09-18 by colinepiloquet