powsybl/powsybl-core #2681: Limits factor Api

EtienneLt, 2023-08-22:
### Describe the current behavior

when computing a security analysis there is nothing to check if the limits (current, voltage,...) are almost overstepped, 
we can have no limit violation when for example the value of the current is 99% of the value of the limit.

### Describe the expected behavior

Add a new API about factor that modify the limit (current, voltage, ...) of an equipment 

# I. Definitions
There should be factors that applied on conditions : (condition can be similar so it will be stated that factors are applied from the beginning to the end)
these conditions are

## 1. factor 
the value we want to applied to the limit (can be over 1 for specific cases, but can not be less than 0 ?)

## 2.  type
it specifies if the factor will be applied to a all branches or to a specified list of them
it should be an enum with values : SPECIFIC, GLOBAL

## 3. brancheIds
if the type (2.) is SPECIFIC it will be the list of the branches the factor will be applied

## 4. situation
it specified if the factor is applied in pre-contingency situation or in post-contingency situation
it should be an enum with values :
- PRE_CONTINGENCY
- POST_CONTINGENCY
- BOTH

## 5. contingencyIds
in case the situation (3.) is POST_CONTINGENCY or BOTH, a list of contingency ids can be specified to indicate on which contingency the factor will be applied

## 6. temporaryLimitsFilter

it should filter on limit with specified time, (if null it is applied on all)
These filter will have some attributes : 

### 6.1. limitsTime 
to specified if the factor is applied on the temporary limits or on the permanent limit or both
it should be an enum with values : TEMPORARY, PERMANENT and BOTH

### 6.2 duration
if the limitsTime (6.1.) is TEMPORARY or BOTH it specified on which temporary limits the factor is applied
it should be a list a of objects the describes duration
the class of the object could have these attributes :
- type (if it is an Equality or a Interval)
- value (only for equality)
- lowBound (only for interval)
- highBound (only for interval)
- lowClosed (only for interval)
- highClosed (only for interval)

# II. concept of code
## 1. LimitFactorList

```java
public class LimitFactorList {
      private List<LimitFactor> factorList;
      private final int version;
      public void apply(Network network);
}
```
## 2. LimitFactor

```java
public class LimitFactor {
      private final double factor;
      private final LimitFactorType type;
      private final List<String> brancheIds;
      private final LimitFactorSituation situation;
      private final List<String> contingencyIds;
      private final TemporaryLimitsFilter temporaryLimitsFilter;

      public void apply(Network network);
}
```

## 3. TemporaryLimitsFilter
```java
public class TemporaryLimitsFilter {
   private LimitsTime limitsTime;
   private List<DurationFilter> filters;
   public boolean filter(TemporaryLimit temporaryLimit) 
}
```

## 4. DurationFilter
```java
public interface DurationFilter {
       public TemporaryLimitsFilterType getType()
       public boolean filter(TemporaryLimit temporaryLimit)
}
```

## 4.1 DurationFilterEquality
```java
public class DurationFilterEquality implements DurationFilter {
       public static DurationFilterType type = EQUALITY; 
       private final double value;
}
```

## 4.2 DurationFilterInterval
```java
public class DurationFilterInterval implements DurationFilter {
       public static DurationFilterType type = INTERVAL; 
       private final double lowBound; 
       private final double highBound;
       private final boolean lowClosed;
       private final boolean highClosed;
}
```

## 5. Json Deserialization/Serialization

the json file will look like : 
```json
{
  "version": "1.0",
  "limitFactors": [ {
      "factor": 0.8,
      "type": "SPECIFIC",
      "brancheIds": [
        "branchId1",        
        "branchId2",
      ],
      "situation": "BOTH",
      "contingencyIds" :  [
           "contingencyId1"
      ],
      "temporaryLimitsFilter": {
        "temporisation" : "BOTH",
        "duration": [
          {
            "type": "EQUALITY",
            "value": 7200
          }, {
            "type": "INTERVAL",
            "lowBound": 1200,
            "highBound" : 99999,
            "lowClosed" : true,
            "highClosed" : false
          }
         ]
      }
    }]
}
```

EtienneLt, 2023-09-26:
to add macro criterium like country voltage level, with exceptions

olperr1, 2023-10-19:
Is replaced by #2751