powsybl/powsybl-core #2697: UCTE importer should allow non active equivalent branches/lines between 2 XNODES

AnkurArohi, 2023-09-12:
### Describe the current behavior

Currently UCTE importer can only build net model when there is no line between 2 XNODES.

Currently there is an exception and model is not built 
Exception is that there is a line between 2 XNODES

### Describe the expected behavior

Actually it is possible to have an equivalent modelling between 2 XNodes.
Considering the documentation there is no where explicitly mentioned that such a model is not considered valid.
Moreover the net should atleast be built with a warning.
It is obvious that the network is then not N-1 conform as there is an equivalent model which will not be directly included in N-1 calculatins but still specially in case one would like to merge this model withthe actual downstream network model below, the equivalent will be replaced a line after merging this network with downstream network so it should function and must be allowed.

### Describe the motivation

After merge the equivalent line would be replaced by the actual downstream network. HEnce the whole network can actually be built.

### Extra Information

If the documentation is followed the 21st character in the Line section has a status field which is set based on integers, if the integer is 9 this means that the model is equivalent and moreover it is out of operation

[image]

AnkurArohi, 2023-09-12:
FYI @annetill

annetill, 2023-09-13:
I can change the exception by a log and ignore the line.