powsybl/powsybl-core #2700: Force CGMES import to bus/breaker topology

annetill, 2023-09-14:
### Describe the current behavior

Sometimes, IGMs in CIM-CGMES are in node/breaker topology. But in these files, we have just a small number of breakers (in general just between bus bar sections). The consequence are multiple. For example, a line contingency with propagation could lead to the loose of many equipments. The line cannot be opened also. In IIDM, this kind of description are typically the bus/breaker topology. 

### Describe the expected behavior

We would like a parameter in CGMES import that forces to import a file in bus/breaker topology. We have a solution (removing the Operation profile) but it is time costly.

jeandemanged, 2023-09-14:
Great idea, also note that in European Merging Function context, EMF is not supposed to recompute topology / rebuild topological nodes. Therefore forcing bus-breaker import is also a valid use case here.

zamarrenolm, 2023-09-25:
Ok to add the parameter.
I assume that you want the bus/breaker level be created from Topological Nodes in the input (TP file must be present, then).

PR #2717 (2023-09-27): CGMES option to force import of node/breaker as bus/breaker
**Please check if the PR fulfills these requirements**

- [X] The commit message follows our guidelines
- [X] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem?**

Fixes #2700 

**What kind of change does this PR introduce?**

**What is the current behavior?**

**What is the new behavior (if this is a feature change)?**
A new parameter has been introduced. Set `iidm.import.cgmes.import-node-breake [... cut]
closed 2023-09-27 by jeandemanged