powsybl/powsybl-core #2736: CGMES export has inf attr fpr r,x for tie lines

AnkurArohi, 2023-10-04:
### Describe the current behavior

Exported CGMES has inf values for the tie lines although the corresponding dangling lines have zero values for these attributes.

### Describe the expected behavior

The tie lines should recieve the sum of the r,x values from the dangling lines, if this leads to issuesin the load flow calculation since they are zero one could add a small value of impedance and make it parameterable.

### Describe the steps

- import network_tso,(with all the BD files)
- import network_dso(with all BD files)
- Merge the networks
- Export the network in CGMES format
- Look for r,x values in the exported CGMES for these tie lines
-

### Environment

pypowsybl 0.22

annetill, 2024-05-15:
@nemanja-st can you check if it is still an issue? It might be easy to create a small network and export it to see if we still have this issue. Thanks!

nemanja-st, 2024-05-16:
> @nemanja-st can you check if it is still an issue? It might be easy to create a small network and export it to see if we still have this issue. Thanks!

Hello @annetill ,
I did a quick test with generating DanglingLine using `DanglingLineNetworkFactory` and setting R and X to 0 and it exported 0:
```java    
    <cim:ACLineSegment rdf:ID="_DL">
        <cim:IdentifiedObject.name>DL</cim:IdentifiedObject.name>
        <cim:ACLineSegment.r>0</cim:ACLineSegment.r>
        <cim:ACLineSegment.x>0</cim:ACLineSegment.x>
        <cim:ACLineSegment.gch>0.0001</cim:ACLineSegment.gch>
        <cim:ACLineSegment.bch>0.00001</cim:ACLineSegment.bch>
        <cim:ConductingEquipment.BaseVoltage rdf:resource="#_100_BV"/>
    </cim:ACLineSegment>
```
I didn't test import/export yet.

annetill, 2024-05-17:
> > @nemanja-st can you check if it is still an issue? It might be easy to create a small network and export it to see if we still have this issue. Thanks!
> 
> Hello @annetill , I did a quick test with generating DanglingLine using `DanglingLineNetworkFactory` and setting R and X to 0 and it exported 0:
> 
> ```java
>     <cim:ACLineSegment rdf:ID="_DL">
>         <cim:IdentifiedObject.name>DL</cim:IdentifiedObject.name>
>         <cim:ACLineSegment.r>0</cim:ACLineSegment.r>
>         <cim:ACLineSegment.x>0</cim:ACLineSegment.x>
>         <cim:ACLineSegment.gch>0.0001</cim:ACLineSegment.gch>
>         <cim:ACLineSegment.bch>0.00001</cim:ACLineSegment.bch>
>         <cim:ConductingEquipment.BaseVoltage rdf:resource="#_100_BV"/>
>     </cim:ACLineSegment>
> ```
> 
> I tested also with:
> 
> * import of test CGMES data and setting r and x to 0 and it was the same result.
> * edit CGMES data to set r and x to 0, import and export and again the same result.
> 
> Conclusion is that I cannot reproduce the issue. If this is still happening, I'd need more information about the model to be able to reproduce

Indeed the issue is the export of a tie line that contains 2 dangling lines with zero impedances.

nemanja-st, 2024-05-31:
> Indeed the issue is the export of a tie line that contains 2 dangling lines with zero impedances.

Hello @annetill,
I managed to reproduce the issue with an older version of pypowsybl, but on the latest version 1.5.1 it's not happening.

@AnkurArohi Can you please update pypowsybl and re-try the export on your side to confirm the issue is not recurring?

AnkurArohi, 2024-06-14:
ok thanks I will try

closed 2024-11-10 by AnkurArohi