powsybl/powsybl-core #2765: Conversion from CGMES to XIIDM loses ID prefix.

ADam-LVA, 2023-11-03:
### Describe the current behavior

Conversion of provided CGMES sample is missing leading prefix "_" in IDs:
<iidm:substation id="87f7002b-056f-4a6a-a872-1744eea757e3" name="Anvers" country="BE" geographicalTags="ELIA-Anvers">
This is true for all RDF IDs.

### Describe the expected behavior

<iidm:substation id="_87f7002b-056f-4a6a-a872-1744eea757e3" name="Anvers" country="BE" geographicalTags="ELIA-Anvers">

### Describe the steps

1. zip all etc/MicroGridTestConfiguration*.xml to MicroGridTestConfiguration.zip
2. Run conversion:
itools convert-network --input-file etc/MicroGridTestConfiguration.zip --output-file MicroGridTestConfiguration --output-format XIIDM

### Environment

Working OK with v4.8.0 (from powsybl-distribution-1.6.0)
Broken from v4.9.0 (from powsybl-distribution-1.7.0)

annetill, 2023-11-03:
You have to use a configuration parameter for CGMES import: network identifiers can be set to rdf:ID (Resource Description Framework Identifiers). RDF identifiers are built from CGMES mRIDs prefixed with a _ character.

The name of the parameter is ````iidm.import.cgmes.source-for-iidm-id````. Its possible values are: `mRID`, `rdfID`. Case is not relevant in the parameter value.

In some business processes, you have to use mRIDs because some other files (remedial actions, glsk, etc.) refer equipments through it. In CGMES 3.0, mRIDs are the real identifiers.

ADam-LVA, 2023-11-03:
Thank you, this explains things.
How to add it to config.yml? Or there is other possibility to use it with itools?

ADam-LVA, 2023-11-04:
I did my homework:
Global configuration in etc/config.yml:
```
import-export-parameters-default-value:
  iidm.import.cgmes.source-for-iidm-id: "rdfID"
```
itools:
`-I iidm.import.cgmes.source-for-iidm-id=rdfID`

closed 2023-11-04 by ADam-LVA