powsybl/powsybl-core #2769: Add reactive power control to ratio tap changers

p-arvy, 2023-11-08:
### Describe the current behavior

The ratio tap changers can only regulate tension.

### Describe the expected behavior

We would like to also control reactive power with ratio tap changers.

p-arvy, 2023-11-08:
I see two possibilities, both in RatioTapChanger class : 
- add an enum (for example RegulationMode) with two values : VOLTAGE and REACTIVE_POWER
- a boolean to say if the regulation is on tension or reactive power

The first option seems better to mimick the enum of PhaseTapChanger class. What do you think @annetill @geofjamg @jeandemanged ?

annetill, 2023-11-08:
Can you put a full description of the API?

p-arvy, 2023-11-08:
Let's say we chose the first option I proposed. The full API of RatioTapChanger class would be :

1. Already existing methods :

double getTargetV();
RatioTapChanger setTargetV(double targetV);
boolean hasLoadTapChangingCapabilities();
RatioTapChanger setLoadTapChangingCapabilities(boolean status);

2. New enum and methods :

enum RegulationMode {VOLTAGE, REACTIVE_POWER};
RegulationMode getRegulationMode();
RatioTapChanger setRegulationMode(RegulationMode regulationMode);
double getTargetQ();
RatioTapChanger setTargetQ(double targetQ);

If we chose the second option, the enum would be replaced by a boolean (e.g. isVoltageRegulation in RatioTapChangerImpl class) and associated methods would be replaced accordingly (e.g. by isVoltageRegulation() and setVoltageRegulation(boolean isVoltageRegulation)).

jeandemanged, 2023-11-09:
For consistency with PhaseTapChanger API I'd go for an enum + regulationValue which can be voltage to reactive power

annetill, 2023-11-09:
And what about generators? We keep this extension or we do similar modifications?

p-arvy, 2023-11-10:
Following morning's discussion, it seems better to continue with enum + regulationValue, as in the PTC class (this implementation was started here #2758 ). 

To avoid making too many changes, it was also decided to leave the targetV method in place, making sure that the control is on voltage. Do we want to set it @Deprecated or not @annetill @geofjamg ?

Finally, to standardise everything, it was proposed that the generator class should also be changed to an enum. What do you think @jeandemanged ?

closed 2024-01-10 by p-arvy