powsybl/powsybl-core #2805: Network modification for connecting/disconnecting a connectable

rolnico, 2023-11-29:
### Describe the current behavior

Connecting/disconnecting a connectable is done by connecting each terminal of the connectable manually.
In case of a connectable with multiple terminals, there is no check that all the terminals can be connected/disconnected before doing it.

### Describe the expected behavior

A network modification could be provided in order to connect or disconnect a connectable. It would check the feasibility on each terminal before doing it.

flo-dup, 2023-12-12:
As discussed, I think the network modifications need to rely on a new disconnect method of the `Connectable` interface:
```java
boolean disconnect(Predicate<Switch> p);
```

Then the implementation on iidm-impl side would be something like this:
```java
public boolean disconnect(Predicate<Switch> p) {
    List<Switch> switchesToDisconnect = new ArrayList<>();
    for (TerminalExt terminal : getTerminals()) {
        if (!terminal.isDisconnectable(switchesToDisconnect, p)) {
            return false;
        }
    }
    switchesToDisconnect.forEach(s -> s.setOpen(true));
    return true;
}
```

The two network modifications will make the link between the corresponding use case (planned outage or forced outage) and the predicate.

flo-dup, 2023-12-12:
For the two network modifications:
- one could be named `PlannedDisconnection` -> opening all kind of switches
- and the other could be named `UnplannedDisconnection` -> opening only breakers kind

Both would have a boolean as parameter to include or not the fictitious switches. By default they should not be included.

flo-dup, 2023-12-12:
For the connection, I think it is enough to have only one `Connection` modification which closes any kind of switches. Maybe a boolean for operating or not fictitious switches is of interest.

closed 2024-01-16 by flo-dup