powsybl/powsybl-core #2880: RemoveFeederBay does not remove linked switches  when a component is removed from a busbreaker model

marqueslanauja, 2024-01-30:
### Describe the current behavior

In `busBreaker` models, when I remove a `ground` component that is connected to the `main bus` throughout a `breaker`, only the `ground` component is removed. The `breaker` and the `bus` where the `ground` was connected remain in the model.

### Describe the expected behavior

The` breaker` and the `bus` where the `ground` was connected have also to be removed.

### Describe the steps

#2879 includes a unitary test reproducing the bug.

Diagram of the model before removing the ground

[image]

Diagram of the model after removing the ground

[image]

annetill, 2024-06-12:
To be confirmed by @flo-dup but I think it is the normal behavior.

flo-dup, 2024-06-19:
yes, this is the "normal" behavior: in bus breaker view, RemoveFeederBay only disconnects the equipment.

Which behaviour would you expect? Similarly to node breaker view, removing all breakers and buses until you encounter a fork or another equipment?

marqueslanauja, 2024-06-19:
> yes, this is the "normal" behavior: in bus breaker view, RemoveFeederBay only disconnects the equipment.
> 
> Which behaviour would you expect? Similarly to node breaker view, removing all breakers and buses until you encounter a fork or another equipment?

Yes. This is the behaviour that I expect.

flo-dup, 2024-09-02:
I confirm that it is normal behaviour, this modification should not remove a bus in bus/breaker view (or, similarly, a busbar section in node/breaker view). Confirmed in technical meeting today.

closed 2024-09-02 by flo-dup