powsybl/powsybl-core #2883: Generic Area model in iIDM

jeandemanged, 2024-02-01:
### Describe the current behavior

We can model Countries and TSOs per Substation

### Describe the expected behavior

We can model any type of area.

TODO: detailled description

### Describe the motivation

Area Interchange Control in LoadFlow and Security Analysis
Zonal automatons

annetill, 2024-05-23:
After a first look at the PR and reading again the issue:
- Why do we need the area type to be an identifiable?
-  Getting the target AC net Interchange (MW, load sign convention, optional) or the net interchange tolerance (MW, optional) is specific to the type of area and I don't understand why we put that in the generic `Area` interface. We need areas for automaton systems, we are not going to put the needed attributes in the same interface. Do I miss something?
- We already have a `Boundary` object: for a good understanding of the iidm model, we must use the same wording. `Point` are no meaning in iidm (maybe in CGMES): we have terminals, nodes, boundaries, buses.
- What is the prupose of public enum BoundaryPointType with HVDC, PAIRED_AC, UNPAIRED_AC and IGNORED_AC ? If the terminal belongs to an HVDC line, it is a DC border, else it is AC. And what means paired, unpaired, ignored? If we have a terminal of a dangling line, it is easy to know if it is paired of not. In which cases it is different ? If it is ignored, why do you need the boundary "point" in the list ?

m-guibert, 2024-05-23:
Hello @annetill, thank you for the feedbacks !
The answers are :

- It is useful to have the Identifiable structure on AreaType-s, with a name, alias, existing serde, etc. Also, using a AreaType instead of a simple String helps ensuring the consistency and unicity of each AreaType
- Those attributes are needed for the AIC computations. They are optional so they won’t impact any other Area usages. Storing them as optional attributes in Area-s seems much cleaner and more practical than storing them in an Area extension.
- The whole "boundaryPoint" model was a WIP and has since been modified. About the paired/unpaired, it can indeed be computed. And ignored elements just won't be added to the Area. So we now use a simpler structure: a BoundaryTerminal record, with just a boolean to indicate if the terminal if on an AC or DC line.

annetill, 2024-05-27:
Just for me, as a reminder of a similar need from our side https://github.com/powsybl/powsybl-core/blob/main/cgmes/cgmes-extensions/src/main/java/com/powsybl/cgmes/extensions/CgmesControlArea.java.

PR #2955 (2024-06-19): Generic Area model in IIDM
**Please check if the PR fulfills these requirements**

- [x] The commit message follows our guidelines
- [x] Tests for the changes have been added (for bug fixes / features)
- [x] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem?**

Fixes #2883 

**Does this PR introduce a breaking change or deprecate an API?**
- [X] Yes
- [ ] No

**If yes, please check if the following requirements are fulfilled**

- [X] The *Breaking Change* or *Deprecated* label has been added
- [X] The migration steps are desc [... cut]
closed 2024-06-19 by olperr1