powsybl/powsybl-core #2885: CGMES export of bus bar section terminals

annetill, 2024-02-01:
### Describe the current behavior

Bus bar sections are present in bus/branch model and in node/breaker model. They have a Terminal. When we export SSH, we expect for all Terminals:
````
<cim:Terminal rdf:about="#_aa31753e-0fc8-11ec-80f5-00155d739404">
        <cim:ACDCTerminal.connected>true</cim:ACDCTerminal.connected>
</cim:Terminal>
````
In updated CGMES export (meaning we keep the original EQ file), we have a reference present in the EQ that is not present in the SSH file. 

### Describe the expected behavior

As iidm model does not have any bus bar sections in bus/breaker topology, we have to keep the id of the bus bar section associated (one or several ?) and in SSH export, we have to export a fake Terminal section with true if the bus is under voltage, else false.

zamarrenolm, 2024-02-02:
We have to be aware that currently when we force the import of node/breaker models as IIDM bus/breaker, we need the TP instance file in the input, and the buses are created based on the Topological Nodes seen in this file.

annetill, 2024-02-02:
> We have to be aware that currently when we force the import of node/breaker models as IIDM bus/breaker, we need the TP instance file in the input, and the buses are created based on the Topological Nodes seen in this file.

The issue raised by @murgeyseb is not related to this feature, even if we could have the issue with it too.

zamarrenolm, 2024-02-02:
I think busbar section terminals can always be considered as connected.

PR #2886 (2024-02-08): CGMES. Updated export of busbar section terminals
**Please check if the PR fulfills these requirements**

- [X] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem?**

Fixes #2885

**Other information**:

When CGMES data is imported to IIDM bus/breaker model the terminals of all busbar sections inside a given CGMES Topological Node are stored in a property of the corresponding IIDM bus. The SSH and TP export mo [... cut]