powsybl/powsybl-core #2999: IIDM 1.13: support of condensers

annetill, 2024-05-02:
### Describe the current behavior

For `powsybl-open-loadflow` and `powsybl-open-reac`, we need a way to make the difference between a generator that is not started and a generator that works as synchronous condenser. It is not possible for the moment, except forcing Pmin and Pmax of the condenser to be equal to zero. 

### Describe the expected behavior

With a new version of xiidm (1.13, 2024.2.0), we propose to add a field `operatingAs` or any proposal with following enum values:
- `GENERATOR` ;
- `SYNCHRONOUS_CONDENSER` ;
- `STATCOM` ; the `operatingAs` field is not really adapted for this case, that it why I am looking for proposals.
- `PUMP`.

jeandemanged, 2024-05-02:
note: CGMES SSH equivalent is `cim:SynchronousMachineOperatingMode` enum with constants:
- `generator`
- `condenser`
- `motor`

annetill, 2024-06-07:
Update: finally, after a talk with @jeandemanged, it seems that a boolean `synchronousCondenser` to true or false is enough to treat the case that a generator is controlling voltage even if it's targetP is equal to zero.

annetill, 2024-06-11:
> Update: finally, after a talk with @jeandemanged, it seems that a boolean `synchronousCondenser` to true or false is enough to treat the case that a generator is controlling voltage even if its targetP is equal to zero.

As I think users will use that field to model STATCOM too (https://www.hitachienergy.com/us/en/products-and-solutions/facts/statcom), even if a good modelling could be to rework on `StaticVarCompensator` later, I think we have to remove "synchronous" from boolean name. Maybe juste "condenser" or "compensator"?

PR #3065 (2024-06-19): Support of condensers
**Please check if the PR fulfills these requirements**
- [x] The commit message follows our guidelines
- [x] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem?**
Yes, fixes #2999

**What kind of change does this PR introduce?**
Feature
closed 2024-06-19 by flo-dup