powsybl/powsybl-core #3023: NullPointerException when exporting SV file in CGMES format

mr0321, 2024-05-17:
### Describe the current behavior

When trying to export SV from imported model in `CGMES `format `NullPointerException `occurs. It happens either: 
1. in all cases for some specific TSO's
2. or there is some specific combination of them

### Describe the expected behavior

Should be able to export the SV in CGMES or should be some understanding why it was not possible (e.g some thresholds were exceeded)

### Describe the steps

1. import the model: `load_from_binary_buffers`: all OK
2. export the model: `save_to_binary_buffer`: exception

No other steps were done between those two

### Environment

- Powsybl version 1.2.1
- Windows 10

### Relevant Log Output

```
java.lang.NullPointerException: null
	at com.powsybl.cgmes.conversion.export.StateVariablesExport$BusTools.getBusViewBus(StateVariablesExport.java:140)
	at com.powsybl.cgmes.conversion.export.StateVariablesExport.writeSvInjectionsForSlacks(StateVariablesExport.java:340)
	at com.powsybl.cgmes.conversion.export.StateVariablesExport.write(StateVariablesExport.java:56)
	at com.powsybl.cgmes.conversion.CgmesExport.export(CgmesExport.java:160)
	at com.powsybl.python.network.NetworkCFunctions.lambda$saveNetworkToBinaryBuffer$10(NetworkCFunctions.java:284)
	at com.powsybl.python.commons.Util.doCatch(Util.java:124)
	at com.powsybl.python.network.NetworkCFunctions.saveNetworkToBinaryBuffer(NetworkCFunctions.java:270)
```

### Extra Information

One possible workaround to solve the issue would be to add additional check to the `cgmes/cgmes-conversion/src/main/java/com/powsybl/cgmes/conversion/export/StateVariablesExport.java` in method `writeSvInjectionsForSlacks` : `Bus bus = st.getTerminal().getBusBreakerView().getBus();` -> `if (bus != null) ` -> `Optional<Bus> optionalBusViewBus = BusTools.getBusViewBus(bus);`

annetill, 2024-05-17:
Thanks for your issue. Before starting a PR to fix this, I have a question in order to understand how you meet this issue. The slack terminal is a terminal connected to the slack bus used by the power flow. At the end of a power flow (open loadflow in fact), the bus cannot be null. But in your case, it seems that the bus connected to the slack terminal is null: the only way is that this terminal has been disconnected. Or we have another issue. Thanks for clarifications!

mr0321, 2024-05-17:
Hi, thank you for your response. This is a strange case indeed. Maybe I am missing something because I am not from this field, but looking through the powsybl code I reached to same conclusion. Working as a part of a team developing a European Merging Function I tested it on the IGM's that I was able to get my hands on (from the OPDM ( ENTSO-E)). It usually occurred with a model from one specific TSO, whereas the validations, loadflow calculations, etc, didn't give any indication that something was wrong. It also happened when using models from several TSO's (merged), in this case, some particular neighbouring TSO pairs or triplets at some timestamps produced the same result (export -> NullPointer, loadflow-> nothing significant to report). Dropping one them usually solved the issue.

annetill, 2024-05-18:
Thanks for these details. We are going to open a PR with your the fix that makes sense of the slack terminal has been disconnected. In the same time, can you bump to https://github.com/powsybl/pypowsybl/releases/tag/v1.4.0 and tell me if you still have the issue on a case ?

closed 2024-05-22 by annetill