powsybl/powsybl-core #3060: Operational limit setters

colinepiloquet, 2024-06-07:
### Describe the current behavior

There are no setters for `TemporaryLimits` which means that to modify one value the whole limit must be recreated. 

### Describe the expected behavior

I think it would we useful to be able to set some of the attributes of the temporary limits, especially the `value`. 

### Describe the motivation

It would be useful to correct faulty data without having to store all the temporary limits and recreate the whole limit.

nao1345678, 2024-07-04:
To correct an already existing limit, we have to make sure the new value fits logically in between the already existing other temporary limits. 
For instance, if the new value is greather than the next one, shouldn't we change the acceptable duration too ? Or should we change the limits' order ?

colinepiloquet, 2024-07-04:
Maybe it's enough to check the other values and if the new limit does not fit logically then the correction would not work ?

olperr1, 2024-08-06:
When a `LoadingLimits` is created, we only log warnings when the temporary limits' values are not consistent (see [`ValidationUtil.checkTemporaryLimits(...)`](https://github.com/powsybl/powsybl-core/blob/4b65c5f45dfdae814350c4e30a2d6d6b1fe4c7f2/iidm/iidm-api/src/main/java/com/powsybl/iidm/network/ValidationUtil.java#L572)).
I think that we can adopt the same principle: allow the inconsistent values, but log warnings in that case.

PR #3113 (2024-08-28): Create setters for temporary limits
**Please check if the PR fulfills these requirements**

- [ ] The commit message follows our guidelines
- [x] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem?**

Fixes #3060

**What kind of change does this PR introduce?**

With this change, we can now change the value of a temporary limit without having to recreate the whole limit. 

**What is the current behavior?**

Temporary limits are immutable, which isn't practi [... cut]
closed 2024-08-28 by olperr1