powsybl/powsybl-core #3152: Add a name property to OperationalLimitsGroup

rcourtier, 2024-09-23:
### Describe the current behavior

At CGMES import, the IIDM OperationalLimitsGroup id is copied from CGMES OperationalLimitSet rdfid. This is neccesary to ensure a strict 1:1 mapping, as well as be able to match limit values update in SSH with limit creation in EQ.
This has however some limitations:
- The rdfid is not human readable, which makes it possible but not easy to later on select the active set.
- The limit set name is stored as a property of the identifiable holding the limits. That way the name can be retrieved during CGMES export but it's not directly accessible in the IIDM OperationalLimitsGroup, which makes it cumbersome to manipulate OperationalLimitsGroup in IIDM.
- The IIDM OperationalLimitsGroup id is by design not globally unique (it is only unique per branch end/transformer leg). Sure it gets unique values during CGMES import since rdfid are globally unique, but there is no guarantee IIDM limits group id are globally unique since they can be created directly within IIDM or from other kind of imports. Because of that, it is not possible to map IIDM OperationalLimitsGroup id to CGMES OperationalLimitSet rdfid at CGMES export.

### Describe the expected behavior

These two changes would facilitate the handling of limits and simplify the CGMES to IIDM mappings:
- Add a new property: "name" to IIDM OperationalLimitsGroup, with a setter and a getter.
- Make the IIDM OperationalLimitsGroup globally unique.

### Describe the motivation

With these changes, it would be possible to:
- Map CGMES OperationalLimitSet rdfid to IIDM OperationalLimitsGroup id, both at CGMES import and export
- Facilitate the selection of the active set:
``` java
Optional<OperationalLimitsGroup> selectedGroup = branch.getOperationalLimitsGroups1().stream().filter(g -> g.getName().equals("WINTER")).findFirst();
selectedGroup.ifPresent(g -> branch.setSelectedOperationalLimitsGroup1(g.getId()));
```

alicecaron, 2025-05-02:
@rcourtier With the work on merge request #3420 we will be able to add properties (such as the name for instance) so maybe we can reuse this and only create the methods to access to the name property if it exists?

alicecaron, 2025-08-07:
The functionnality of this issue is covered by #3420 and #3504: no new name attribute is added in the operational limits group related equipement but directly on the OperationalLimitsGroup new populated properties: `CGMES.OperationalLimitSetName` and `CGMES.OperationalLimitSetRdfID`.