powsybl/powsybl-core #3191: Export to UCTE format

alicecaron, 2024-10-24:
### Describe the current behavior

According to the documentation [here](https://powsybl.readthedocs.io/projects/powsybl-core/en/latest/grid_exchange_formats/index.html) Poswybl cannot export to UCTE for an imported file in other format than UCTE.

### Describe the expected behavior

Powsybl can convert a network not in UCTE format to an UCTE formatted file.

### Describe the motivation

As CGMES is not yet used everywhere as entry file, some applications need to be able to export a DIE network to an UCTE.

### Extra Information

- generate special ids for UCTE (to support the [UCTE format](https://eepublicdownloads.entsoe.eu/clean-documents/pre2015/publications/ce/otherreports/UCTE-format.pdf) rules)
- need a default naming strategy
- some equipments are not supported and muste be implemented:
  - ThreeWindingsTransformer to model by 3 TwoWindingsTransformer => 1 node, 1bus, 1voltage level, 1 substation, 3 2WT
  - HvdcLine not supported => 1 group?
  - StaticVarCompensator => 1 group?
  - DanglingLine ? => 1 half line, 1load, 1bus, 1voltage level, 1substation
  - other?

- voltage regulation rules
  - see the UCTE format documentation
  - ? todo: to describe
  - ?

- update the [doc](https://powsybl.readthedocs.io/projects/powsybl-core/en/latest/grid_exchange_formats/index.html), in particular the array for the supported conversion formats

So-Fras, 2024-12-16:
This issue is partly fixed by the #3206 PR. The mapping of new elements is still to be done.

DanielMuellerTBW, 2026-09-17:
@So-Fras @alicecaron could you please comment on this topic once again? Are there currently any efforts to provide Powsybl with a generic, broadly applicable UCTE export? If not, what are the reasons behind that? From what I can see, the related PRs appear to be quite advanced already. Why was the final step towards implementation not taken?

jeandemanged, 2026-09-18:
> [@So-Fras](https://github.com/So-Fras) [@alicecaron](https://github.com/alicecaron) could you please comment on this topic once again? Are there currently any efforts to provide Powsybl with a [...]

Hello @DanielMuellerTBW,
cc @So-Fras , @alicecaron ,
In open-source, implementation happen only when someone is doing the work and/or is funding it.
One TSO had the need described in #3989, and is as far as I know the only one currently supporting this effort. My company is in charge of the implementation for their needs, progressively detailed as sub-issues of #3989. This will be presented at the next [PowSyBl Technical Steering Committee meeting](https://www.powsybl.org/pages/community/regular-meetings.html) Oct 1st, 2026. TSC meetings are open to anyone, feel free to join and contribute by sharing your use case and issues.

PR #3206 (2024-12-16): More robust UCTE naming strategy 
**Please check if the PR fulfills these requirements**

- [x] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem?**

Fixes #3191 

**What kind of change does this PR introduce?**

**What is the current behavior?**

**What is the new behavior (if this is a feature change)?**

**Does this PR introduce a breaking change or deprecate an API?**
- [X] Yes
- [ ] No

**If yes, please  [... cut]