powsybl/powsybl-core #3236: Unexpected error when locking out an equipment

pgambier, 2024-12-02:
### Describe the current behavior

When locking out a line with a "fork" on one side with a disconnector (see image attached), there is an un expected error.
[image]

### Describe the expected behavior

The highlighted switch should open.

flo-dup, 2024-12-10:
What do you think should occur when calling `connect()` for a fork @pgambier? if the disconnector of the other line and the bbs switch are opened.
With a diagram, that may be clearer; with the voltage level below, if calling connect on line L2, should the breaker only be closed after, or should also the line L1 be connected?
[image]

pgambier, 2024-12-10:
I think we should split the issue and fix the disconnect part only in this one

closed 2024-12-10 by flo-dup