powsybl/powsybl-core #3292: How to convert CGMES triples into a powsybl Network

bishoyroufael, 2025-01-28:
Hi!

After a lot of researching, I couldn't find a way to go from a triple store to a powsybl `Network`. I don't want to read the a file directly from powsybl. The workflow will be that triple store will be queried from an _Apache Jena_ server and then fed into powsybl for load flow calculations.

Here's what I got:
```java
  public static void main(String[] args) {
        // Step 1: Create a Jena Dataset and load RDF data
        // Model model = ModelFactory.createDefaultModel();
        
        TripleStoreJena tripleStore = new TripleStoreJena();
        // Load RDF data from a file or resource
        try (InputStream inputStream = Main.class.getResourceAsStream("/SmallGridTestConfiguration_BC_EQ_v3.0.0.xml")) {
            if (inputStream == null) {
                throw new IllegalArgumentException("RDF data file not found!");
            }
            tripleStore.read(inputStream, "","eq.xml");
            // RDFDataMgr.read(model, inputStream, Lang.TURTLE);
        } catch (Exception e) {
            e.printStackTrace();
        }
        
        // How to generate a network from the triple store?
        // Network cgmesNetwork = Network.read()

        System.out.println("Done!");
    }
```
I am on version `6.6.1`. Can you please guide me?

bishoyroufael, 2025-07-30:
Is there any updates on this? Does PowSyBl support Apache Jena as a triple store as of the time being? I have a use-case where I want to run load-flow calculations on data stored in Apache Jena

alicecaron, 2025-08-05:
Hello @bishoyroufael ! Sorry for the delay. Today, Powsybl does not support Apache Jena as a triple store.
However it would indeed be interesting to have this possibility in the future. We currently do not have time to allocate to this specific feature.
If you want you can contribute to code ; we would be glad to review it. We can also discuss about technical solutions in this issue.