powsybl/powsybl-core #3318: A network with mixed topology kind (node-breaker and bus-breaker) should be exported to CGMES as bus-branch.

rcourtier, 2025-02-10:
### Describe the current behavior

In PowSyBl, the topology kind is defined for each `VoltageLevel`, and it's perfectly fine to have overall in the network a mixed topology kind, that is have some `VoltageLevel` in `NODE_BREAKER` and some other `VoltageLevel` in `BUS_BREAKER`.

With the current CGMES exporter, when _at least_ one `VoltageLevel` is in `NODE_BREAKER`, the whole network is considered a `NODE_BREAKER` and the connectivity is exported as follow:
* `NODE_BREAKER` `VoltageLevel`'s adjacent `node`s of the NodeBreakerView will be exported as CGMES `ConnectivityNode`s
* `BUS_BREAKER` `VoltageLevel`'s `Buses` of the BusBreakerView will be exported as CGMES `ConnectivityNode`s

UML-wise, the export is correct and passes the quality checks. However, this is misleading since what is exported is not the detailed connectivity (for bus-breaker voltage levels) but the topology disguised as connectivity.

### Describe the expected behavior

The CGMES export of a network with a mixed topology kind should be a bus-branch export: when _at least_ one `VoltageLevel` is in `BUS_BREAKER` topology kind, no `ConnectivityNode` should be exported, only the `TopologycalNode` from the `Buses` of the BusBreakerView.

zamarrenolm, 2025-02-11:
Be aware that with this approach we will lose information in the export. The parts of the network that were specified as node/breaker will be simplified in the output.

Also, for CGMES 3, even exporting a bus/breaker model requires writing CGMES connectivity nodes. Exporting a bus/breaker model using base connectivity designed for node/breaker is valid, and seems the approach chosen by CGMES.

rcourtier, 2025-03-05:
> Be aware that with this approach we will lose information in the export. The parts of the network that were specified as node/breaker will be simplified in the output.
>
> Also, for CGMES 3, even [...]

This is handled in https://github.com/powsybl/powsybl-core/pull/3315. In case of a CIM 16 export, we will indeed by default lose the detailed connectivity in the export. But there is a simple way not to: explicitely request a node-breaker export by setting a parameter (added in 3315) so that the connectivity is exported.
As for CIM 100 export, the node-breaker export will be the default export type for hybrid networks so by default no information will be lost.

closed 2025-03-17 by rcourtier