powsybl/powsybl-core #3341: PSS/E import failed in case of out of service badly connected elements

geofjamg, 2025-03-04:
### Describe the current behavior

The issue could be reproduced with a modified version of IEEE 14 test case

```
0 / END OF GENERATOR DATA, BEGIN BRANCH DATA
    1,     2,'1 ', 0.01938, 0.05917,0.05280,   0.00,   0.00,   0.00,  0.00000,  0.00000,  0.00000,  0.00000,1,1,   0.0,   1,1.0000,   0,1.0000,   0,1.0000,   0,1.0000
    1,     5,'1 ', 0.05403, 0.22304,0.04920,   0.00,   0.00,   0.00,  0.00000,  0.00000,  0.00000,  0.00000,1,1,   0.0,   1,1.0000,   0,1.0000,   0,1.0000,   0,1.0000
    2,     3,'1 ', 0.04699, 0.19797,0.04380,   0.00,   0.00,   0.00,  0.00000,  0.00000,  0.00000,  0.00000,1,1,   0.0,   1,1.0000,   0,1.0000,   0,1.0000,   0,1.0000
    9999,  4,'1 ', 0.05811, 0.17632,0.03400,   0.00,   0.00,   0.00,  0.00000,  0.00000,  0.00000,  0.00000,0,1,   0.0,   1,1.0000,   0,1.0000,   0,1.0000,   0,1.0000
```
The last branch is connected on from side to a 9999 bus which does not exist, BUT the status is 0 which means the branch is out of service.
PSS/E tool allows this, we had the case on a real file and it was perfectmy imported in PSS/E.
PowSybl failed at validation step with such a message:
```
[main] WARN com.powsybl.psse.model.pf.PsseValidation - NonTransformerBranch: Unexpected I: 9999
[main] WARN com.powsybl.psse.model.pf.PsseValidation - PSS/E Validation end. ValidCase false
```

geofjamg, 2025-03-04:
From the real file 
```
WARN  11:37:23.260 j.u.ArrayList        | NonTransformerBranch: Unexpected J: 3007
WARN  11:37:23.260 j.u.ArrayList        | NonTransformerBranch: Unexpected I: 3552
WARN  11:37:23.266 j.u.ArrayList        | Transformer: Unexpected I: 3007
WARN  11:37:23.266 j.u.ArrayList        | Transformer: Unexpected J: 34475
WARN  11:37:23.266 j.u.ArrayList        | Transformer: 3007 34475 2  Unexpected winding1 Cont: 34475
```
It seems we also have an issue with the winding CONT too.

zamarrenolm, 2025-03-04:
We propose to check what PSSE does in these situations and try to do the same to fix the bad input data before completing the import.

A priori, for the line without the right references, either we discard it or keep it in the model. If we keep it we need to create a fictitious bus.

For the transformer missing the regulated bus (CONT) either we deactivate the regulation or try to assign it to one of the ends of the transformer.

PR #3345 (2025-03-06): Psse: discard badly connected equipment and fix badly defined controlled buses
**Please check if the PR fulfills these requirements**

- [x] The commit message follows our guidelines
- [x] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] A PR or issue has been opened in all impacted repositories (if any)

**Does this PR already have an issue describing the problem?**

Fixes #3341

**What kind of change does this PR introduce?**

Bug fix

**What is the current behavior?**

Psse models with badly connected equi [... cut]
closed 2025-03-06 by geofjamg