powsybl/powsybl-core #3384: PSS/E zero resistance HVDC failed to import

geofjamg, 2025-03-26:
### Describe the current behavior

When importing "TWO-TERMINAL DC DATA" section we failed in case of zero resistance HVDC line.
PSS/E simulator seems to handle this case correctly without error.

### Describe the expected behavior

We should be able to import zero resistance HVDC lines as PSS/E simulator does.

pjanecek0, 2025-03-26:
Same problem with Non transformer branch with x=0. PSS/E replaces X with a small value.

So-Fras, 2025-07-02:
This is the original feedback from the user: 

> In PSS/E, there are two types of HVDC, "two-terminal DC" (representing LCC) and VSC DC. The first type seems to be imported correctly. The problem lies with VSC-type links.

PR #3500 (2025-07-07): Psse: Unit test importing a two-terminal DC line and a VSC, both with zero resistance
**Please check if the PR fulfills these requirements**

- [x] The commit message follows our guidelines
- [x] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] A PR or issue has been opened in all impacted repositories (if any)

**Does this PR already have an issue describing the problem?**

Fixes #3384
A unit test has been added importing a two-terminal DC and a VSC, both with zero resistance.
In the current version of the i [... cut]