powsybl/powsybl-core #3400: Improve Network reading methods

alicecaron, 2025-04-07:
### Describe the current behavior

To import/read a network into an IIDM `Network` a lot of parameters muste be set.

### Describe the expected behavior

To import/read a network into an IIDM `Network` we can only precise the source to the network and only the wanted parameters to override (none if need be).

### Describe the motivation

For a lot of use cases the use of` Network::read` does not need a lot of parameters. The mandatory parameter should always be source to get the network data to read (currently a `String` with an `InputStream`, a `Path` or a `ReadOnlyDataSource`). But to simplify the reading of a network the other parameters could be set by default and overriden only if needed (`ComputationManager`, `ImportConfig`, `Properties`, `ImportersLoader`, `ReportNode`, `NetworkFactory`).

For that we can create for instance a `NetworkReaderBuilder` with a builde mecanism that sets all the parameters by default and allows them to be overriden.

### Extra Information

Need to chose if we keep all the the current methods or not because if we remove them or some of them there will be a breaking change.

olperr1, 2025-04-09:
I don't think we should introduce a `NetworkReaderBuilder` class/interface. If we follow the builder pattern, it is expected to have a `build()` method creating a `NetworkReader`. Then using it would be something like:
```java
Network.newNetworkReaderBuilder("path/to/network.xiidm")
    .withComputationManager(manager)
    .withReportNode(report)
    .build()
    .read();
```

Instead, we can directly introduce a `Reader` inner class/interface with default values and setters:
```java
Network.newReader("path/to/network.xiidm")
    .setComputationManager(manager)
    .setReportNode(report)
    .read();
```

geofjamg, 2025-04-09:
> I don't think we should introduce a `NetworkReaderBuilder` class/interface. If we follow the builder pattern, it is expected to have a `build()` method creating a `NetworkReader`. Then using it would be something like:
> 
> Network.newNetworkReaderBuilder("path/to/network.xiidm")
>     .withComputationManager(manager)
>     .withReportNode(report)
>     .build()
>     .read();
> Instead, we can directly introduce a `Reader` inner class/interface with default values and setters:
> 
> Network.newReader("path/to/network.xiidm")
>     .setComputationManager(manager)
>     .setReportNode(report)
>     .read();

Agree. Also we should keep existing Network.read() for simple cases.