powsybl/powsybl-core #3511: Model a PQU diagram for reactive limits

alicecaron, 2025-07-09:
### Describe the current behavior

Generators, dangling lines and battery at least have reactive limits. These reactive limits can be of two types currently: 

- `MinMaxReactiveLimits`: determined by a minQ and maxQ
- `ReactiveCapabilityCurve`: determined by a minQ and maxQ per given P (active power)

### Describe the expected behavior

Generators, dangling lines and battery at least have reactive limits. These reactive limits can be of two types currently: 

- `MinMaxReactiveLimits`: determined by a minQ and maxQ
- `ReactiveCapabilityCurve`: determined by a minQ and maxQ per given P (active power)
- a new object determined by a minQ and maxQ per given P per given U (voltage)

### Describe the motivation

Model a volume of reactive limits for a generator for instance, depending on both the voltage and the active power.

### Extra Information

Create a new `ReactiveLimits` implemenation allowing to compute minQ and maxQ according to given active power AND voltage. This may imply a refactor of the `ReactiveLimits` implementations + modify the iidm, allowng to define new kinds of reactive limit (a volume instead of only a reactive capability curve)

snoopfab, 2025-10-03:
Can we agree that the volume expected must be a single convex polyhedron (or polytope) ?

The provided example code, which checks if a point satisfies a conjunction of linear inequalities, correctly defines such a region. 

```
public boolean isInside(double p, double q, double v) {
    for (Plane plane : listOfPlanes) {
        double expression = q + plane.alpha * v + plane.beta * p;
        if (plane.relation == '<' && expression >= plane.gamma) {
            return false;
        }
        if (plane.relation == '>' && expression <= plane.gamma) {
            return false;
        }
    }
    return true;
}

```
Given this simple, constraint-based definition, using a complex geometry library may be unnecessary overkill for our current usage.

We probably should implement checks to detect invalid or degenerate inputs, which should trigger an error:

- Input constraints that define multiple, disconnected convex regions (as this violates the definition of a single convex polyhedron).
- An empty feasible region (e.g., conflicting constraints like two parallel planes pointing in opposite directions.

Concerning  the scope of the: 

limited to modelling Java + IIDM serialisation / deserialisation , validity checks and utility method to test if a p,v,q point is in the reactive limit envellope ? Or we should go further (usage in rao / openloadflow ?)

snoopfab, 2025-10-08:
Thanks for your insights.

I was thinking to use apache common math3 that implements a simple primal simplex algorithm with slacks variables and infeasibility detection. 
This would allow us to detect both degenerate inputs (when lp is infeasible we throw an empty polytope exception to the user). 
By the way, this is already an existing dependency of powsybl math (from pom.xml):
```
<dependency>
    <groupId>org.apache.commons</groupId>
    <artifactId>commons-math3</artifactId>
</dependency>
```

>Why ? we can simply fixe the voltage to the current calculated voltage (the bus v attribute) if defined of fallback to the nominal voltage so that we always come back to a full (p, v) detection.

We could use the linear solver to minimize and maximize Q under the feasible polytope region to compute them without needing to fix the voltage. But i don't know if this is really needed by olf or somewhere else.

snoopfab, 2025-10-16:
@alicecaron @geofjamg 

Concerning the q + alpha * v + beta * p < OR > gamma equations there's no way to simply bound v and p independently from q as q has an implicit coefficient fixed to one. 
Should'nt we let the user configure optionnal bounds on each variables without specifying hyperplanes ?

geofjamg, 2025-10-16:
> [@alicecaron](https://github.com/alicecaron) [@geofjamg](https://github.com/geofjamg)
>
> Concerning the q + alpha * v + beta * p < OR > gamma equations there's no way to simply bound v and p [...]

There is already p (pmin, pmax on generator element) and v bounds (low_voltage_limit and high_voltage_limit in the voltage levele element) in the IIDM model